HKUST-Aerial-Robotics/Teach-Repeat-Replan #3: Will it support A3 flight controller?

SiuKeungm, 2019-07-05:


USTfgaoaa, 2019-07-05:
@SiuKeungm 
The controller supports A3 perfectly. 
But the intermediate layer, djiros, may have some issues. It depends on the version of the onboard SDK (OSDK). As far as I know, OSDK version < 3.3 suits it. But for a new A3 with the latest firmware (it corresponds to OSDK 3.7 or 3.8), it may not work.

So, the conclusion is, the 'n3ctrl' is good, but the 'djiros' may not be good. But I believe it can be fixed by slightly modify some interface.

ziv-lin, 2019-07-05:
Hey, you can type `git checkout 3.3` to fix the compile errors. 
Shown as following~
[image]

closed 2019-07-06 by USTfgaoaa

SiuKeungm, 2019-07-06:
@USTfgaoaa @ziv-lin Thanks for your reply. I will try it!