HKUST-Aerial-Robotics/Teach-Repeat-Replan #4: The plugin for class 'rviz_plugins/Goal3DTool' failed to load.

huangwen0907, 2019-07-18:
Hi, Nice project, I love this `Teach-Repeat-Replan`.
Now I followed your advices and install all the files and it can run. 
But there are some errors which i can not control it.  I am very appreciate that 
 someone can help me .
- 1 The plugin for class 'rviz_plugins/Goal3DTool' failed to load. The errors show as below:
[image]

- 2 The [local_replanner-1] process has died, show as below:
[image]
- 3 The depthImage's status is 'warn', Maybe this one is not a question, Just for your information
[image]

- 4 hhahah, the dirty words.
[image]

I am a new beginner in this project, so forgive the Idiot question

ZbyLGsc, 2019-07-18:
@huangwen0907 The reason you can not load the Goal3DTool cloud be not sourcing the setup.bash before running the simulation(source setup.bash in the same terminal you launch the simulation). As for the dying of local_replanner, I can not figure out what happened from the single screenshot. More details will help, please.

huangwen0907, 2019-07-18:
@ZbyLGsc Thank you for your reply, follow your step, I find out that  the reason  of `dying of local_replanner`. As before I use the `zsh` which change my environment, now I switch to the `bash` and the node died do not happen again. 
However, it still load the ` the Goal3DTool cloud ` faild.
[image]

- One more question. I can not reply the path when I press the button `M`or (auto) mode when I finish the manual mode(flying a circle without any obstacle). It print the info that `it can not find a safe path`(show as below)
[image]

[image]

ZbyLGsc, 2019-07-18:
@huangwen0907 You can ignore 1,3,4 safely. As for problem 2, I just try the simulation several times and can not reproduce your result. More details will be helpful.

huangwen0907, 2019-07-18:
@ZbyLGsc ok, Thank you very much. 
Could you please give me some reply about the `replay` can not be executed when I enter the `auto` mode ?

ZbyLGsc, 2019-07-18:
@huangwen0907 Congrat! And I already removed the above mentioned plugins, since they are unused in this project.

USTfgaoaa, 2019-07-18:
Hi bro,

1. The dependencies on rviz plugin has no harm to this program. And boyu has fixed them

2. According to our paper，we find the topo equivalent path of the teaching. So if you fly the drone a circle without go around any obstacle，the topo equivalent path is nothing. So try to teach the drone to fly around obstacles and then go back（just like the demo in our git），and press M again.

huangwen0907, 2019-07-18:
@USTfgaoaa  Thank you .  All the works is valid now except onething, It can be control in manual mode and a new replay path  can be generated in auto mode. However when I entered the auto mode, the simulation vehicle landed on the ground automatically. So the vehicle did not do the replay work(fly with the generation replay path). This is normal? Show as below:
[image]

huangwen0907, 2019-07-18:
@ZbyLGsc  Nice! Good Catch.  If the bug exists, in the real world, when armed the motors, the vehicle may inverted, This is very dangerous.

-  5) it show the error of my licesense.
> Error MSK_RES_ERR_MISSING_LICENSE_FILE - 'A license cannot be located.'
[image]

-6) In my rviz, no depthIMage is showed. could you tell me the topic is published from which node?
[image]

USTfgaoaa, 2019-07-18:
Hi，

1. You should approve a academic license in Mosek website. Follow the instruction in ReadMe.

2. In CPU mode，depth image will not be published. If you want it, turn on GPU mode in the local sensing package. But firstly you should make sure you have installed CUDA properly. Please follow the ReadMe for details.

huangwen0907, 2019-07-18:
@USTfgaoaa  OK, I will have a try. Thank you.
All my problems is solved now, I will close this issues. If you have any question, please feel free to reopen this.

closed 2019-07-18 by huangwen0907

lei0907, 2020-12-30:
How did you solve the first problem?The plugin for class 'rviz_plugins/Goal3DTool' failed to load. If you know, please tell me how to solve it.Thank you very much.

VamosCC, 2021-09-17:
> How did you solve the first problem?The plugin for class 'rviz_plugins/Goal3DTool' failed to load. If you know, please tell me how to solve it.Thank you very much.

After 'catkin_make' the pkgs, which include rviz_plugins with Goal3DTool, You should excute '**source devel/setup.sh**', then rviz can locate to current project defined rviz_plugins with Goal3DTool, and the question solved.