HKUST-Aerial-Robotics/Teach-Repeat-Replan #5: Can not pilot the drone using my keyboard and errors occured when the system launched by ./trr_simulation.sh

KangchengLiu, 2019-07-27:
Hi, Gaofei and Boyu,
Nice project and I like this Teach-Repeat-Replan.
Now I followed your advices and install the packages and the package can be launched. But I can't pilot the drone using my keyboard. And some errors occured and I have no idea how to deal with them.
[image]
1. The joystick force feedback can not be opened.
[image]
2. 2 processes have died.
[image]
[image]
 I will appreciate if you can help me, thanks!

USTfgaoaa, 2019-07-27:
@KangchengLiu 
Hi, I haven't encountered problems with joystick before, but I will try to help you fix it.
May you comment this line 'rosrun joy joy_node & sleep 0.2;' in the trr_simulation script, and try launch it again to see whether the process die?

USTfgaoaa, 2019-07-27:
@KangchengLiu 
Do you connect a joystick to your computer?

KangchengLiu, 2019-07-27:
It finally turned out that the process died because the sleep time between roslaunch trr_global_planner and roslaunch trr_global_planner was too short that my laptop can not deal with it.

You can fix it by doing roslaunch one by one or increase the sleep time in the trr_simulation script.

Thanks Gao Fei for his kind help!

closed 2019-07-27 by KangchengLiu