HKUST-Aerial-Robotics/Teach-Repeat-Replan #6: sdf_map publishing issue

ChengeYang, 2019-08-01:
Hi,

I am buidling a drone local planner based on the sdf_map module of your project. I build the local_map package and import the sdf_map as library.

I try to use it in the simplest way 
1. In cpp code I call the class as below:
```
SDFMap sdf_map;
sdf_map.initMap(nh_);

ros::Rate loop_rate(10);
while (ros::ok())
{
      sdf_map.publishMap();
      sdf_map.publishMapInflate();
      sdf_map.publishUpdateRange();
      sdf_map.publishESDF();

      ros::spinOnce();
      loop_rate.sleep();
}
```

2. In launch file, I make the node listen to depth image topic and camera pose topic and get the published pointcloud from topic /sdf_map/occupancy, which  is then visualized in rviz.
However, the published map seems to be responding very slowly (<10Hz), and is limited to a local area. The local region published from /sdf_map/update_range is giving the right region, which is around the drone.

The first figure shows that the drone is stading on the ground. At this status, the pointcloud can be published crrectly.
But in the second figure, when the drone takes off and get away from the ground, the bounding box seems to be located correctly, but the pointcloud is not rising along with the drone.

[image]

[image]

Do you have any idea about this issue? Any help will be highly appreciated.

USTfgaoaa, 2019-08-01:
@ChengeYang 

I guess it may be related to the size of the map? We limit the height of the map as 3m in this project.
You can check it.

ChengeYang, 2019-08-01:
Is the map size set by ROS parameters "sdf_map/global/x_size", "sdf_map/global/y_size", "sdf_map/global/z_size" in launch file?

By the way, is this local planner always maintaining a global map stored in "occupancy_buffer", but only considering the local one when planning a local path?

ZbyLGsc, 2019-08-01:
@ChengeYang 
1. yes.
2. the occupancy_buffer can store a map up to the size of sdf_map/global/*_size. When receiving an image and pose pair, only the local map in the red bounding box is updated, due to the limited sensing range.
3. the "cut_height" is used to limit the height of the visualized point cloud (points whose z > cut_height will not be published). So I think you can increase this value and check again.

ChengeYang, 2019-08-01:
Thanks for the reply!
I increased the cut_height to 10, but in rviz it seems that the pointcloud is still cut at a given height, as in the figure below. Although it's better than before, but still not high enough for me. Any idea?

ChengeYang, 2019-08-01:
OK I solved it by increasing "sdf_map/global/z_size" and "sdf_map/cut_height" to a large number at the same time.

But in my PC the map seems very easy to crash if I set the map size to be big although my memory is not fully used.

Thanks for the help!

ChengeYang, 2019-08-01:
One more question, when do the planning, do you plan within the whole global map, etc. use the occupancy buffer, or only use the local map which is the red bounding box showed in rviz?

USTfgaoaa, 2019-08-02:
@ChengeYang 

Can you test your program and find where it crashed? To find whether it's a memory, you can try to reduce the x-y size of the map, or set a larger resolution, to see whether it crash again.

For the latter question, I think you mean the onboard local re-planning, at that time only the local map shown in the bounding box is used for planning.

USTfgaoaa, 2019-08-03:
@ZbyLGsc 
Have any idea about the crash issue?

ZbyLGsc, 2019-08-03:
@ChengeYang I just reproduced the crash described by you. The reason is that some points exceed the global boundary of the map. You can try the latest commit now.

USTfgaoaa, 2019-08-03:
@ChengeYang Please close this issue if it's solved.

ChengeYang, 2019-08-03:
Yep I''ll close it now.

Also, the sdf_map in your fast_planner also has the similar issue. Please check.

@ZbyLGsc @USTfgaoaa  Thanks for the help!

closed 2019-08-03 by ChengeYang