HKUST-Aerial-Robotics/Teach-Repeat-Replan #7: Is it possible to make the sdf_map moving with the robot?

ChengeYang, 2019-08-08:
Hi,

Thanks for the help last time! I've got another problem regarding sdf_map.

I'm trying to fly the drone in a big environment. It requires the map_size to be really big and takes too much memory. So I'm thinking to make the occupancy_buffer moving with the robot instead of keeping still. In this way, the map size can be significantly reduced.

Do you have any idea how can I do it?
Thanks.

USTfgaoaa, 2019-08-09:
@ChengeYang 

Your problem is just about whether to keep or forget the historical map of what the drone explored. If you don't want to keep a large global map data buffer, just maintain a smaller buffer along with the drone's movements for local planning, you should keep fusing and cleaning the data on your local map and calculate the ESDF of this local map every tick you want to update it. You may take Ewok(https://github.com/VladyslavUsenko/ewok) as a reference.

The change of the frame_id has no effect but only affects the visualization in Rviz.

ChengeYang, 2019-08-11:
I'll close this issue now. Thanks!

closed 2019-08-11 by ChengeYang