HKUST-Aerial-Robotics/Teach-Repeat-Replan #8: Controller problem

M1nHuan, 2019-08-28:
We use A3 flight controller and I have installed DJI_ROS 
DO I need to install N3Ctrl ?

USTfgaoaa, 2019-08-28:
@M1nHuan 
Yes.
The N3Ctrl works as the onboard controller to control the drone's position and velocity. 
The attitude of the drone is controlled by the autopilot. 
DJI_ROS is just an intermediate software for the communication between the onboard controller and the N3/A3 autopilot.

M1nHuan, 2019-08-29:
Can it(N3Ctrl pkg) work on A3 flight controller ？

M1nHuan, 2019-08-29:
And when I git clone https://github.com/HKUST-Aerial-Robotics/N3Ctrl.git , it needs username and password for github. Then I entered my own account ,but it not work.
How can I deal with this problem ?

USTfgaoaa, 2019-08-29:
@M1nHuan 
1. For the compatibility issue, I think it can not perfectly suit A3, you have to check interfaces and modify the code accordingly.
2. I check out this problem and found that now the N3Ctrl.git repo doesn't exist anymore. We will soon modify the ReadMe.
So you should directly clone the Teach-Repeat-Replan repo and switch to 'experiment' branch, then you will find N3Ctrl under the controller folder.

closed 2019-09-01 by USTfgaoaa