HKUST-Aerial-Robotics/Teach-Repeat-Replan #9: eigen assertion failed.

Doodle1106, 2019-08-29:
Hi guys,

I am runing it without GPU, and every time I use 

`
./trr_simulation.sh`

eigen internal check will fail.  I can reproduce it on another machine running ubuntu 16.04, ROS Kinetic and eigen 3.3.7.
Following is the complete log relate to this issue:

`... logging to /home/ss/.ros/log/df172d28-ca17-11e9-8e48-9cb6d0e498fb/roslaunch-ss-XPS-15-9560-10785.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ss-XPS-15-9560:46837/

SUMMARY
 

PARAMETERS
 * /RouterNode_1/input: /position_cmd_input
 * /cascade_control_1/acceleration_coeff: [0.1, 0.0, 0.0, 0...
 * /cascade_control_1/acceleration_limit: [50, -50, 0.1, 50...
 * /cascade_control_1/corrections/p: 0
 * /cascade_control_1/corrections/r: 0
 * /cascade_control_1/corrections/z: 0
 * /cascade_control_1/gains/ang/x: 0.13
 * /cascade_control_1/gains/ang/y: 0.13
 * /cascade_control_1/gains/ang/z: 0.1
 * /cascade_control_1/gains/rot/x: 1.5
 * /cascade_control_1/gains/rot/y: 1.5
 * /cascade_control_1/gains/rot/z: 1.0
 * /cascade_control_1/mass: 0.98
 * /cascade_control_1/position_coeff: [3.3, 0.0025, 0.0...
 * /cascade_control_1/position_limit: [50, -50, 0.1, 50...
 * /cascade_control_1/quadrotor_name: test
 * /cascade_control_1/simulation: True
 * /cascade_control_1/velocity_coeff: [2.0, 0.0005, 0.0...
 * /cascade_control_1/velocity_limit: [50, -50, 0.1, 50...
 * /odom_visualization_1/base_frame: /map
 * /odom_visualization_1/color/a: 1.0
 * /odom_visualization_1/color/b: 0.0
 * /odom_visualization_1/color/g: 0.0
 * /odom_visualization_1/color/r: 1.0
 * /odom_visualization_1/covariance_scale: 100.0
 * /odom_visualization_1/drone_scale: 0.8
 * /odom_visualization_1/id: 1
 * /odom_visualization_1/target_frame: /map
 * /odom_visualization_1/tf45: True
 * /odom_visualization_1/tfc: False
 * /quadrotor_simulator_so3_1/rate/odom: 100.0
 * /quadrotor_simulator_so3_1/simulator/init_state_x: -8.5
 * /quadrotor_simulator_so3_1/simulator/init_state_y: -8.5
 * /quadrotor_simulator_so3_1/simulator/init_state_z: 0.5
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /state_machine_node_1/drone_id: 1
 * /state_machine_node_1/horizontal_gain: 1.2
 * /state_machine_node_1/joymap_back: 6
 * /state_machine_node_1/joymap_pitch: 4
 * /state_machine_node_1/joymap_roll: 3
 * /state_machine_node_1/joymap_start: 7
 * /state_machine_node_1/joymap_thrust: 1
 * /state_machine_node_1/joymap_yaw: 0
 * /state_machine_node_1/mission/button_A: RouterNode
 * /state_machine_node_1/mission/button_B: RouterNode
 * /state_machine_node_1/mission/init_mission: 2
 * /state_machine_node_1/simulation: True
 * /state_machine_node_1/vertical_gain: 1.2
[... 480 lines cut ...]

process[local_replanner-1]: started with pid [11716]
process[time_optimal_traj_server-2]: started with pid [11717]
MAX Velocity := 1
MAX Acceleration := 2.5
local_replanner: /usr/local/include/eigen3/Eigen/src/Core/DenseStorage.h:128: Eigen::internal::plain_array<T, Size, MatrixOrArrayOptions, 32>::plain_array() [with T = double; int Size = 4; int MatrixOrArrayOptions = 0]: Assertion `(internal::UIntPtr(eigen_unaligned_array_assert_workaround_gcc47(array)) & (31)) == 0 && "this assertion is explained here: " "http://eigen.tuxfamily.org/dox-devel/group__TopicUnalignedArrayAssert.html" " **** READ THIS WEB PAGE !!! ****"' failed.
Stack trace (most recent call last):
#8    Object "[0xffffffffffffffff]", at 0xffffffffffffffff, in �
#7    Object "/home/ss/software/TRR/devel/lib/trr_local_replanner/local_replanner", at 0x417e28, in _start
#6    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7ff26e44182f, in __libc_start_main
#5    Object "/home/ss/software/TRR/devel/lib/trr_local_replanner/local_replanner", at 0x41586b, in main
#4    Object "/home/ss/software/TRR/devel/lib/trr_local_replanner/local_replanner", at 0x425751, in SDFMap::SDFMap()
#3    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7ff26e44ec81, in __assert_fail
#2    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7ff26e44ebd6, in 
#1    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7ff26e458029, in abort
#0    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7ff26e456428, in gsignal
Aborted (Signal sent by tkill() 11716 1000)
check_collision_time_interval:= 0.1
time_commit := 0.1
time_horizon:= 4
time_lead_yaw:= 1
pub_lead_yaw:= 1
[local_replanner-1] process has died [pid 11716, exit code -6, cmd /home/ss/software/TRR/devel/lib/trr_local_replanner/local_replanner ~replan_mission:=/time_optimal_traj_server/check_pts /sdf_map/depth:=/pcl_render_node/depth /sdf_map/pose:=/pcl_render_node/camera_pose /sdf_map/odom:=/drone_1/visual_slam/odom /sdf_map/cloud:=/pcl_render_node/local_pointcloud __name:=local_replanner __log:=/home/ss/.ros/log/df172d28-ca17-11e9-8e48-9cb6d0e498fb/local_replanner-1.log].
log file: /home/ss/.ros/log/df172d28-ca17-11e9-8e48-9cb6d0e498fb/local_replanner-1*.log
`

USTfgaoaa, 2019-08-29:
@Doodle1106 
1. Did you change some parameters in the launch file?
2. Did you miss something when configuring this repo? Since no one has reported such an 'Eigen crash' issue so far.
I will test the latest commit of this repo and reply to you soon.

USTfgaoaa, 2019-08-29:
@Doodle1106 
I check the log you post, I think it's a memory issue: '[polyhedron_generator]{CPU} finish cpu cluster
*** Error in `/home/ss/software/TRR/devel/lib/trr_global_planner/tr_node': free(): invalid next size (normal): 0x00000000020aac40 ***'

However, I test the code and I can not reproduce it on my computer. So first let me know if you have changed some parameters, or just use everything by default.

USTfgaoaa, 2019-08-29:
@Doodle1106 
I don't know if it's because you have limited memory space. I reduce the default memory allocated. You can pull the newest code and try again.

Doodle1106, 2019-08-29:
I pulled the lastest code and I can reproduce this error on both machines: 

`local_replanner: /usr/local/include/eigen3/Eigen/src/Core/DenseStorage.h:128: Eigen::internal::plain_array<T, Size, MatrixOrArrayOptions, 32>::plain_array() [with T = double; int Size = 4; int MatrixOrArrayOptions = 0]: Assertion `(internal::UIntPtr(eigen_unaligned_array_assert_workaround_gcc47(array)) & (31)) == 0 && "this assertion is explained here: " "http://eigen.tuxfamily.org/dox-devel/group__TopicUnalignedArrayAssert.html" " **** READ THIS WEB PAGE !!! ****"' failed.
Stack trace (most recent call last):
#8    Object "[0xffffffffffffffff]", at 0xffffffffffffffff, in �
#7    Object "/home/gishr/software/TRR/devel/lib/trr_local_replanner/local_replanner", at 0x417e28, in _start
#6    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f47a301082f, in __libc_start_main
#5    Object "/home/gishr/software/TRR/devel/lib/trr_local_replanner/local_replanner", at 0x41586b, in main
#4    Object "/home/gishr/software/TRR/devel/lib/trr_local_replanner/local_replanner", at 0x425751, in SDFMap::SDFMap()
#3    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f47a301dc81, in __assert_fail
#2    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f47a301dbd6, in 
#1    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f47a3027029, in abort
#0    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f47a3025428, in gsignal
Aborted (Signal sent by tkill() 27401 1000)`

They are both on ubuntu 16.04, Ros Kinetic, boots 1.58. Judging by abovementioned error I think it might be related to incompatible Eigen as I am using eigen 3.3.7 on both machines. Also there might be a chance that "EIGEN_MAKE_ALGIGNED_OPERATOR_NEW" is missing inside certain class but I haven't had the time to check it. What version of Eigen are you guying using? I will give it a try and get back to you asap.

USTfgaoaa, 2019-08-29:
@Doodle1106 
I checked out my Eigen version. It's 3.3.9

Doodle1106, 2019-09-10:
solved by installing on a fresh new environment. Thanks!

closed 2019-09-10 by Doodle1106