HKUST-Aerial-Robotics/Teach-Repeat-Replan #11: Some problems in the experimental part

DYYYYYYYYY, 2019-09-25:
Hello, I installed the code for the experimental part on the drone and the ground station as you said. However, when I run   ./sensing_estimation.sh  , it always appears the prompt the prompt for 
       wait for imu             (I have changed the location of realsense_stereo_imu_config.yaml to the actual path), how is this going?

USTfgaoaa, 2019-09-26:
@DYYYYYYYYY 

In the default setting we use the IMU from DJI autopilot (N3) instead of Realsense D435i. Because RealSense's IMU is much worse and may cause VIO failure. 

Your problem is related to the localization part, the VINS VIO we used in this project. If you want to use VINS with realsense's IMU, you should first modify its configuration to receive the IMU data from realsense driver (which bridges realsense) instead of djiros (which bridges N3).

DYYYYYYYYY, 2019-09-26:
Thank you for your answer, but I just used the topic /djiros/imu, but it always reminds me: wait for imu. I echoed /djiros/imu topic, it has data. But why has it always been prompted: wait for imu? Is it not subscribed to the topic?

USTfgaoaa, 2019-09-26:
@DYYYYYYYYY Yes, maybe. You may check VIN's configuration, to see whether it listens to /djiros/imu.

DYYYYYYYYY, 2019-09-29:
Thank you for your explanation, this issue has been resolved. 
There is still a problem. I found that the installation direction of the N3 flight controller on your drone is different from the normal situation. What impact will this have? If there is an impact, where do I need to modify the parameters?

USTfgaoaa, 2019-09-30:
@DYYYYYYYYY That doesn't matter. But you need to configure the right direction in DJI assistance.

closed 2019-10-03 by USTfgaoaa