HKUST-Aerial-Robotics/Teach-Repeat-Replan #13: some problem in experiment part

M1nHuan, 2019-09-29:
您好，我们想请问在地面站启动mapping 和 机载端启动sensing 之后，应该怎么去操作无人机去建图？
我们之前端着无人机在实验室走了几圈，结果建图的效果并不理想，建图的时候是否有什么特殊技巧？

USTfgaoaa, 2019-09-30:
@M1nHuan  没有什么特殊技巧，但是这个跟系统里很多因素有关。首先确保你的定位准确，loop closure及时。之后调试realsense得到较好的深度图。

M1nHuan, 2019-09-30:
非常感谢您的回复
我们在建图的时候，地面站出现的画面很卡，经常出现未响应的情况
请问这个跟地面站的计算能力有关还是与无线传输有关？
请问你们的地面站配置是什么？

USTfgaoaa, 2019-09-30:
@M1nHuan 
The ground station we use is Alienware with i7+GTX1080 or Hasee with i7+GTX1060 GPUs. However the mapping phase doesn't use any GPU resources but may require a good CPU.

Both of these two issues may be related to your problem.
I suggest while debugging you can buy some very long Ethernet cables and  to use these wired connections instead of the WIFI to test the mapping.

M1nHuan, 2019-10-03:
Thanks for your apply！
We use wired connections and the performance of mapping has been improved

closed 2019-10-03 by USTfgaoaa