HKUST-Aerial-Robotics/Teach-Repeat-Replan #14: some question in experiment part

M1nHuan, 2019-10-11:
Dear author：
  We meet some problem in experiment part.
  Q1：In teaching phase ，after teaching the drone a expected path ， I  pressed m button，but the drone dose not fly as my expected path in rviz. Do I need press m button ? or Should I directly run ctrl_estimation_replanning.sh in the onboard computer after teaching ?
 Q2: In the process of mapping, because the wireless delay is too high, we use the wired connection, but the actual flight can not rely on the wired connection, we want to know the wireless network equipment you use.
  I‘ll be very grateful for your reply！

USTfgaoaa, 2019-10-11:
@M1nHuan 

1. For real world experiments, actually the drone is triggered by a button on the remote controller. As you can see, even if the drone fly in simulation, it has nothing related to the real-world. But these guidances (how to trigger the drone by a real drone's remote controller) are missing in this repo currently.

2. In the repeating phase, once the global planning is done and the drone starts to fly, nothing is done on the off-board ground-station, so don't worry about the wireless connection. During repeating, the ground station is only used to visualization.

By the way, recently my students and I are also re-distributing this system, of course in ZJU since I have graduated from my group in HKUST. We also found a lot of missing/misleading guidance should be added in this github repo. Also, some guys reported difficulties in replicating this project.

My student Zhou Xin @bigsuperZZZX  will upload a primary tutorial, fix some ambiguous instructions and possible bugs in this weekend. And we will add more details later in next week. We're also considering adding some Chinese instructions. Wish that would help you a lot.

USTfgaoaa, 2019-10-11:
@DYYYYYYYYY 
1. Of course yes. I think we clearly write that during repeating, the drone operates with full autonomy, which means no ground station is used.
2. Of course that depends on a lot, such as the speed of the drone (time to react, motion blur in images) and the quality of your depth measurements (fine tune the realsense camera, tune the local map and local planner's parameters). So it depends on how "greatly" you change the obstacles. Do you think the changes in the demo video is greatly enough?

USTfgaoaa, 2019-10-11:
@M1nHuan @DYYYYYYYYY 

Hey guys, I'm happy to see others replicating our project, although it's quite a complicated, big, inter-dependent project. So how about we build a WeChat group and discuss? Third-party's applications can also help us refine this project or find something to dig into for research purpose.

closed 2019-10-12 by USTfgaoaa