HKUST-Aerial-Robotics/Teach-Repeat-Replan #16: About mapping

tlb9551, 2019-10-15:
Thank you for your open source work.
Does this code contain the part of the mapping module (from Image-Pose pair to Depth Image)? 
We are trying to generate dense depth maps online with a monocular camera and VINS-mono as VIO, do you have any open source code recommended?

USTfgaoaa, 2019-10-15:
@tlb9551 No it doesn't.

Your problem is not about mapping, bu dense depth estimation. More specifically, you need motion stereo because you wanna a monocular system. You may try VI-MEAN (https://github.com/HKUST-Aerial-Robotics/VI-MEAN), REMODE, open_quadtree_mapping (https://github.com/HKUST-Aerial-Robotics/open_quadtree_mapping).

tlb9551, 2019-10-15:
Thank you for your reply！
Best wishes for you.

closed 2019-10-15 by tlb9551