HKUST-Aerial-Robotics/Teach-Repeat-Replan #20: IMU Frequency Setting for Matrice 100

ArlenCHEN, 2019-10-31:
Hi Dr. Gao,

Firstly thanks so much for your this excellent open source work!

I encountered a problem when I tried to run the code on Matrice 100 (M100). 

I am wondering if this code is compatible with M100? If yes, then as [http://wiki.ros.org/dji_sdk](url) shows, M100's imu frequency is only 100 Hz while other platforms have 400 Hz as required by VINS-Fusion. Any suggestion on how to change the code to allow an IMU frequency of 100 Hz? BTW, I am also wondering if the controller in this code support M100 N1 controller?

Thanks!

Best,
Zheng

USTfgaoaa, 2019-10-31:
@ArlenCHEN 

Firstly, this controller is designed for M100's N1 controller originally, and we modify it to suit N3/A3. So, I think you can try it on M100 and fix some interfaces bugs if necessary.
Secondly, I don't think the IMU 100/400 Hz is an issue. Both frequencies should be OK for VINS-Fusion.

ArlenCHEN, 2019-10-31:
@USTfgaoaa 

Thanks for your kind reply!

With published topics `/camera/infra1/image_rect_raw`, `/camera/infra2/image_rect_raw`, and `/dji_sdk/imu`, I started the vins node, however I constantly received the prompt:

> wait for imu ... 

Also, the estimation of vins drifted hugely. 
With the given config file in the code, I changed the `left.yaml` and `right.yaml,` but I didn't change the `estimate_extrinsic` between IMU and camera. Is it possible for the reason could be the bad initial guess of `estimate_extrinsic`?

Thanks!
Zheng

USTfgaoaa, 2019-10-31:
@ArlenCHEN  wait for imu ... means it did't get IMU msgs properly. So check the pipeline first. I suppose you don't use the djiros package I provided. Since I find the /dji_sdk/imu in your question. No we don't use the official DJI SDK, we use a customized version dji_ros.

ArlenCHEN, 2019-10-31:
I tried `roslaunch djiros djiros.launch` in workspace, however I got this error:

> STATUS/1 @ activate, L1290: Activation successful
[ INFO] [1572533996.780187844]: drone activated
[djiros-1] process has died [pid 21576, exit code -11, cmd /home/zheng/catkin_trrexp/devel/lib/djiros/djiros_node __name:=djiros __log:=/home/zheng/.ros/log/c0413b9e-fbd8-11e9-b96c-48a472d9b408/djiros-1.log].
log file: /home/zheng/.ros/log/c0413b9e-fbd8-11e9-b96c-48a472d9b408/djiros-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete

But with the official DJI SDK and with `roslaunch dji_sdk sdk.launch`, I got the dji node started.

ArlenCHEN, 2019-10-31:
It seems the code breaks here(`WORKSPACE/src/controller/djiros/src/modules/dji_sdk_node.cpp`):
[image]

USTfgaoaa, 2019-11-01:
@ArlenCHEN There must be some interfacing issues, since the djiros I released is suited for N3/A3 not for N1. Sorry I haven't encountered this issue and I can't debug it for you. You may check djiros here: https://github.com/groundmelon/djiros.
 Or you continue to use the official sdk, but pay attention to the msgs pipeline.

ArlenCHEN, 2019-11-03:
@USTfgaoaa 
Thanks so much for your suggestion. I will check it out soon and post the solution here.

ArlenCHEN, 2019-11-05:
Solved.

The problem is due to the incompatibility between M100 controller and the modified djiros package. Specifically, it's about how to read telemetry data. There are two ways to read the telemetry data: broadcast and subscription, see [https://developer.dji.com/onboard-sdk/documentation/guides/component-guide-telemetry.html](url)

M100 only supports telemetry broadcast while the new versions(branch A3 and 3.3) of the modified djiros only support subscription. To solve this, just simply switch to branch 3.1: [https://github.com/groundmelon/djiros/tree/3.1](url)

Another way is to use dji official sdk as [USTfgaoaa](https://github.com/HKUST-Aerial-Robotics/Teach-Repeat-Replan/issues/20#issuecomment-548633105) pointed. But just remember to change the imu topic to `/dji_sdk/imu` in config file.

Thanks so much for your suggestion! @USTfgaoaa

closed 2019-11-05 by ArlenCHEN