HKUST-Aerial-Robotics/Teach-Repeat-Replan #21: Huge drift of VINS-Fusion

ArlenCHEN, 2019-11-05:
With IMU on DJI Matrice 100 and Intel Realsense D435i camera, I have a huge drift of Vins-Fusion like this:
[image]
I guess the reason could be the wrong parameters (camera-IMU-extrinsics, IMU noise, IMU random walk and shift time between camera and IMU). 
To obtain those parameters, I have some ideas:

1. camera-IMU-extrinsics: Manually measure the transition distance and rotation relation between camera and IMU.
2. IMU noise and IMU random walk: Generate Allan standard deviation charts.
3. shift time: Adopt the given value in the code (18ms).

I am working on those things but I am not sure if they are correct. Can anyone give some hints? Thanks!

tlb9551, 2019-11-07:
1. Do you close the IR projector of D435i camera? (**important**)
2. A suggestion: Using the D435i's imu is not a good idea unless you have hardware damping on it, the vibration of the propeller may make the VIO work poorly.

ArlenCHEN, 2019-11-07:
@tlb9551 
Thanks for your suggestion!
1. You mean the IR stereo camera of D435i? I can obtain the two infra image topics and actually I can obtain the normal tracker images of VIO.
2. I did not use D435i's IMU but the one on DJI Matrice 100. 

I have obtained the IMU intrinsics after generating the Allan deviation charts. But I am not sure how to obtain the camera-IMU-extrinsics since I don't know the exact position and coordinate of IMU in N1 controller.

tlb9551, 2019-11-07:
@ArlenCHEN According to my understanding of the D435(i) whitepaper, the D435 uses an infrared projector to project infrared spots to the front in order to enhance the performance of depth estimation in flat areas. If the infrared projector is not disabled, the feature tracker will always track these spots instead of the environmental features. So I suggest you check the output of the feature_tracker.

ArlenCHEN, 2019-11-07:
@tlb9551 
Thanks very much for your advice! It indeed works better now.
But I still have a transformation problem between body frame and world frame. With the provided cam_imu_extrinsics in `realsense_stereo_D435i_imu_config.yaml` (I suppose this is an acceptable guess for the initial cam_imu_extrinsics since I have not measured the extrinsics on my device yet), I have an initial camera pose like this:
[image]
However, without doing anything, the pose of camera could rotate like this:
Firstly:
[image]
Next:
[image]
And then:
[image]

Then I try to tilt and move my device a little bit. The camera pose could drift like this:
[image]

Then I found the online estimated extrinsics parameters during the above move and put them in config file, the results are worse (Of course it should be, since I didn't see the right relation during testing).

Any idea? Thanks!

USTfgaoaa, 2019-11-08:
@ArlenCHEN 
Yes @tlb9551 's suggestions are correct. You should first close the emitter in realsense and forget about imu in realsense. 
Then your problem is about how to "use" the vio and how to do the spatial and temporal calibration between the camera and IMU. VINS-FUSION has integrated both these two functionalities. You should turn on these functions and move the device to excite the calibration. Or you may try some tools for calibration, such as Kalibr.

ArlenCHEN, 2019-11-11:
@USTfgaoaa 
Yes, the performance is better after closing the emitter. But when I move my device, it still can drift a lot even I turn on the online calibration. I guess the reason is the bad initial "guess" for the imu-camera-extrinsics. I adopt **body_T_cam0**, **body_T_cam1**, and **center_T_imu** in `realsense_stereo_D435i_imu_config.yaml` as the extrinsics parameters, which are perhaps very different from my device. I think providing a good initial "guess" could probably solve the problem. So could you share how you guys compute the initial guess for those imu-camera-extrinsics? 
Thanks!

USTfgaoaa, 2019-11-14:
@ArlenCHEN OK, for mu setup, I just measure it manually for the initial guess. I think you may also try Kalibr to calibrate it offline as the initial guess.

ArlenCHEN, 2019-11-20:
@USTfgaoaa 
Thanks for your advice. I manually measure the initial guess for M100 and have a good results now! One thing to note is to initialize the VINS with Intel RealSense D435i and 100Hz IMU on M100, slow and translational movement is preferred. The position and scale estimation look good now. 

However, I cannot have a body frame aligned with world frame. There is always a misalignment between body frame and world frame. See below pic. It shows the initialized body frame is misaligned with world frame. The system is put on ground and I fixed the world frame. When moving the system, the movement is consistent with body frame but inconsistent with world frame. I guess one reason could be a wrong initialization. Another possible reason could be the IMU inside M100 is not installed in an axis-aligned-with-world way. I am trying to figure out the reason.
[image]

Anyone can give some hints? Thanks!

closed 2019-12-19 by ArlenCHEN