HKUST-Aerial-Robotics/Teach-Repeat-Replan #22: About estimating hover thrust online.

tlb9551, 2019-11-07:
When using the **THRUST** command to control the N3 flight control to achieve the desired **FORCE** (always oriented toward the +z-axis of the body frame), there is always a problem that how to find the mapping between the thrust and the force.
I noticed that in your code, you try to use a class **HovThrKF** to estimate hover thrust online. 
HovThrKF::update() use the Kalman filter, but this function is never used.
(https://github.com/HKUST-Aerial-Robotics/Teach-Repeat-Replan/blob/7900aa660831b5b766eb7898c0dbcc30a7bee009/onboard_computer/controller/n3ctrl/src/N3CtrlFSM_control.cpp#L95)
HovThrKF::simple_update() use a simple filter, and the relationship between thrust and force is considered to be proportional. Means:  F = c * thrust. Is this based on some facts?
I have ever worked on PX4 (Pixhawk), and I used to think that thrust is proportional to rotor speed w, and force is proportional to w^2, so the thrust is proportional to sqrt(force).

bigsuperZZZX, 2019-11-13:
Yes we consider force is proportional to thrust.
If you read the code in function _Controller::update_ in [Teach-Repeat-Replan/onboard_computer/controller/n3ctrl/src/controller.cpp](https://github.com/HKUST-Aerial-Robotics/Teach-Repeat-Replan/blob/experiment/onboard_computer/controller/n3ctrl/src/controller.cpp), you will notice that we use a simple proportional model as well.

tlb9551, 2019-11-13:
@bigsuperZZZX 
Thank you for your reply, I understand now.
All is well.

closed 2019-11-13 by tlb9551