HKUST-Aerial-Robotics/Teach-Repeat-Replan #24: Maybe a tiny mistake in experiment code

SIRwkp, 2019-12-10:
In the function `void pubLatestOdometry()` from TRR/onboard_computer/localization/VINS-Fusion/vins_estimator/src/utility/visualization.cpp, there exists

`w_R_center = Q.toRotationMatrix() * center_R_imu;`

I think it should be 

`w_R_center = Q.toRotationMatrix() * center_R_imu.transpose();`

Am I right?

SIRwkp, 2020-01-07:
1、 I fix the first problem by adding a few lines in function void rcvJoyCallBack() of file teach_repeat_planner.cpp, like this: 

---TRR-SIM/global_planner/src/teach_repeat_replanner.cpp
+++TRR-SIM/global_planner/src/teach_repeat_replanner.cpp
@@ -147,156  +147,159  @@
    else if(joy.buttons[6] == 1.0)
    {
        ROS_WARN("[trr_global_planner] Enter in maunal flight mode");
        clearVisualization(_manual_path.size(), _vis_iter_num);

        _manual_path.clear();
        _polyhedronGenerator->reset();
        _has_traj = false;
        _vis_iter_num = 0;
+quadrotor_msgs::SpatialTemporalTrajectory space_time_msgs;
+space_time_msgs.action = quadrotor_msgs::SpatialTemporalTrajectory::ACTION_ABORT;
+_space_time_pub.publish(space_time_msgs);
    }

2、I'm sure I'm right about question two, hope that could be modefied.

3、And I find another problem that the height of the drone will drop down in manual mode in simulaiton with only horizontal joystick input without vertical joystick input. I fix it like this:

---TRR-SIM/state_machine/src/rc.cpp
+++TRR-SIM/state_machine/src/rc.cpp
@@ -430,439  +430,159  @@
    if (hostRC->hasYawControl)
    {
      cmd_pos.yaw_dot = hostRC->joy.sticks[STICK_YAW] * hostRC->yawGain;
      Eigen::Quaterniond q;
      q.x() = hostRC->odom.pose.pose.orientation.x;
      q.y() = hostRC->odom.pose.pose.orientation.y;
      q.z() = hostRC->odom.pose.pose.orientation.z;
      q.w() = hostRC->odom.pose.pose.orientation.w;
-      vel   = q.toRotationMatrix() * vel;
+Eigen::Vector3d vel_tmp = q.toRotationMatrix() * vel;
+vel(0) = vel_tmp(0);
+vel(1) = vel_tmp(1);
    }

I will close the issue. Hope it's helpful to others.

closed 2020-01-07 by SIRwkp

USTfgaoaa, 2020-01-08:
@SIRwkp thanks for your suggestions!
@bigsuperZZZX Xin, please check this issue afterwards.