HKUST-Aerial-Robotics/Teach-Repeat-Replan #27: Odometry for controller

ArlenCHEN, 2020-01-04:
@USTfgaoaa 
May I ask why do you use `/vins_estimator/imu_propagate`, instead of `/vins_estimator/odometry` as the odometry for the controller? Thanks!

USTfgaoaa, 2020-01-05:
@ArlenCHEN For a higher feedback rate.

ArlenCHEN, 2020-01-05:
Thanks! But I am wondering is there any way to change the frequency of `/vins_estimator/odometry` to meet the control feedback rate in VINS_FUSION?

USTfgaoaa, 2020-01-05:
@ArlenCHEN No way to change that msgs, which is exactly the frequency of the pose estimation of the VIO. It's impossible to boost the frequency of the VIO to 50~100 Hz in an onboard computer.

ArlenCHEN, 2020-01-05:
Clear now. Thanks for your kind reply!

closed 2020-01-05 by ArlenCHEN

gouyiyong, 2020-02-03:
> @ArlenCHEN For a higher feedback rate.
@USTfgaoaa  are they /vins_estimator/odometry and /vins_estimator/imu_propagate much different?no any difference for control performance? and I wanna make sure if  /vins_estimator/imu_propagate is accurate enough for the controller and the whole system.