HKUST-Aerial-Robotics/Teach-Repeat-Replan #28: odometry

gouyiyong, 2020-02-03:
are they /vins_estimator/odometry and /vins_estimator/imu_propagate much different? no any difference for control performance? and I wanna make sure if /vins_estimator/imu_propagate is accurate enough for the controller and the whole system. is it possible to run the controller with /vins_estimator/odometr 10 Hz? I read this paper https://onlinelibrary.wiley.com/doi/full/10.1002/rob.21732, you are using 10Hz odometry for the controller.

USTfgaoaa, 2020-02-03:
@gouyiyong  
1) We (our lab, past and now) never use 10Hz pose feedback to control the drone. If anyone (or paper) states 10 Hz, it's wrong. Also, I think we state in that paper we use 100 Hz pose feedback, clearly.

[image]

2) As I have explained,  /vins_estimator/odometry is the pose in the sliding window of the visual-inertial estimator, but /vins_estimator/imu_propagate is with the IMU integration.
3) /vins_estimator/imu_propagate is accurate, very accurate, as long as your VINS runs properly and smoothly.
4) I think 10 Hz is far not good, based on my experience. And you can try it. Of course, you can use 10 Hz feedback if the performance satisfies your expectations.

gouyiyong, 2020-02-03:
> @gouyiyong
> 
> 1. We (our lab, past and now) never use 10Hz pose feedback to control the drone. If anyone (or paper) states 10 Hz, it's wrong. Also, I think we state in that paper we use 100 Hz pose feedback, clearly.
> 
> [image]
> 
> 1. As I have explained,  /vins_estimator/odometry is the pose in the sliding window of the visual-inertial estimator, but /vins_estimator/imu_propagate is with the IMU integration.
> 2. /vins_estimator/imu_propagate is accurate, very accurate, as long as your VINS runs properly and smoothly.
> 3. I think 10 Hz is far not good, based on my experience. And you can try it. Of course, you can use 10 Hz feedback if the performance satisfies your expectations.

@USTfgaoaa  thanks for your detailed explanation, got it :)

closed 2020-02-03 by gouyiyong