HKUST-Aerial-Robotics/Teach-Repeat-Replan #29: DJI N3 imu frequency

gouyiyong, 2020-02-04:
Hi,

is it possible to lower or higher the imu frequency of the DJI N3? until now I don´t find any method in DJI web. sometimes the vins-mono occurs this error ¨imu... disorder¨ and I am wondering if it is caused by a too high frequency of the imu data. if lowering the frequency, will the low-level controller performance be influenced?
DJI N3 controller can trigger one camera with a changeable frequency by rosservice call different frequency or some other method. I test 10 Hz works fine, but 20 Hz drifts a lot, looks really weird.

Best,

USTfgaoaa, 2020-02-27:
@bigsuperZZZX please check it.

bigsuperZZZX, 2020-02-28:
1. You can only change your imu frequency within some enum numbers {0, 1, 10, 50, 100, 200, 400} at [line 73 in DjiRosPublisher.cpp](https://github.com/HKUST-Aerial-Robotics/Teach-Repeat-Replan/blob/experiment/onboard_computer/controller/djiros/src/djiros/DjiRosPublisher.cpp), because N3 flight controller has a fixed control period with 2.5ms. It is notified somewhere in OSDK source code I've seen before, but I am failed to find it out again this time. 
2. Changing this value too low will do bad effect on [n3ctrl](https://github.com/HKUST-Aerial-Robotics/Teach-Repeat-Replan/tree/experiment/onboard_computer/controller/n3ctrl), but won't affect the control performance of N3 flight controller.
3. Imu error from VINS always cased by overlarge clock deviation between imu and image, maybe adjusting time offset can help. This value is called "[ts](https://github.com/HKUST-Aerial-Robotics/Teach-Repeat-Replan/blob/experiment/onboard_computer/localization/VINS-Fusion/config/realsense/realsense_stereo_D435i_imu_config.yaml)" in our project.
4. DJI N3 can not work well with Intel RealSense D435i using hardware synchronization during our tests.

USTfgaoaa, 2020-02-28:
@gouyiyong check the above comment , plz

gouyiyong, 2020-03-02:
> @gouyiyong check the above comment , plz
@USTfgaoaa @bigsuperZZZX  thanks a lot

closed 2020-03-03 by gouyiyong