HKUST-Aerial-Robotics/Teach-Repeat-Replan #30: Magnetic field interference

ArlenCHEN, 2020-02-26:
Dear authors @USTfgaoaa @bigsuperZZZX,  

I noticed a problem when I fly the drone in an area with strong magnetic interference. The yaw behavior is weird even the pose(position and orientation) is provided by VICON or VINS. It seems the drone always uses its compass regardless of the sources of localization information. Is this true? Could you kindly give me some hints? Thanks!

USTfgaoaa, 2020-02-27:
@ArlenCHEN yes it's true. I can't solve your problem since I always used the default yaw control of the autopilot. If you want to change its feedback to external pose estimation, I guess you have to check the manual of onboard SDK.

ArlenCHEN, 2020-02-28:
Thanks! I will post the solution if I can find one.

ArlenCHEN, 2020-03-03:
Sorry to bother this issue again. @USTfgaoaa @bigsuperZZZX 
Does yaw_rate_control could be used in this situation? I noticed in yaw_rate_control mode, u_yaw, instead of des.yaw is published out. So does that mean in this way the default yaw control in autopilot is disabled? Thanks!

USTfgaoaa, 2020-03-04:
@bigsuperZZZX

bigsuperZZZX, 2020-03-04:
Sorry I don't know how the autopilot behaves in this situation. Maybe more tests have to be conducted to determine it.

closed 2020-04-10 by ArlenCHEN