HKUST-Aerial-Robotics/Teach-Repeat-Replan #31: Trajectory Generation Fails in Real Experiment after Teaching

ArlenCHEN, 2020-03-07:
Hi @USTfgaoaa @bigsuperZZZX,

I built a map using a real drone. Then in the simulator I controlled the drone model to generate a series of convex hulls. However, when I pushed the auto mode button on joystick, there was always an error, shown as below:
[image]

According to one previous similar issue, I taught the drone around obstacles, see below:
[image]
Nothing happened when entering the auto mode but just with the above error. No idea how this happens. Please suggest. Thanks!

ArlenCHEN, 2020-03-07:
Sometimes the result is just messed up, as below:
[image]

bigsuperZZZX, 2020-03-08:
@ArlenCHEN We have encountered these problems as well, but they are rare. Make sure that the start and the end point of the teaching trajectory are not too close to the ground. If that doesn't helps, maybe you can lower the map size and change the resolution of the map. If all these trials failed, send your map to me please(_iszhouxin@zju.edu.cn_), I will try to fix it.

ArlenCHEN, 2020-03-08:
@bigsuperZZZX Thanks! I will try them and let you know the results.

ArlenCHEN, 2020-03-09:
@bigsuperZZZX 
I tried to change the resolution. It works now.
The original resolution is 0.15. It turns out that when the resolution is within [0.34, 0.38], the optimization works for my case. Also, the trajectory result varies with the change of `optimization/min_order` (of course). For the purpose of validation, I list the results with the `min_order` from 1 to 4 (I run the node 4 times and of course the four teaching trajectories are different, but I control them with a slight difference. Below the resolution is 0.34.).
`min_order` = 1.0:
[image]
`min_order` = 2.0:
[image]
`min_order` = 3.0:
[image]
`min_order` = 4.0:
[image]

For those who fail to run the optimization, I believe enlarging the map size might be helpful since the map size limits the convex hulls size. If you look at the above case with `min_order`=3, you can find the convex hull should be large enough to contain the highly jerky optimized result (if this is what you want), otherwise, the optimization would fail. 

@bigsuperZZZX For your convenience to check, I already sent you the .pcd file of my map. Thanks!

ArlenCHEN, 2020-03-09:
@bigsuperZZZX @USTfgaoaa 
Sorry authors, I have another question about the trajectory. 

The `tr_node` in the experiment code publishes the trajectory with a message type of `quadrotor_msgs/SpatialTemporalTrajectory`. However, the n3ctrl controller listens to commands which have a message type of `quadrotor_msgs/PositionCommand`. Do you include any conversion into the code? I cannot find such a function. Thanks!

bigsuperZZZX, 2020-03-10:
@ArlenCHEN Thank you very much for these tests, which can surely help us improve the performance of trajectory generation.
To your latest question, [**here**](https://github.com/HKUST-Aerial-Robotics/Teach-Repeat-Replan/blob/master/local_replanner/local_replanner/src/traj_server.cpp) is the module for that job, which your can find easily by `rqt_graph`. The _traj_server(trajectory server_) gets the global trajectory from the `tr_node` and keeps receiving the newest local trajectory from the local replanner and outputs position command.

ArlenCHEN, 2020-03-10:
@bigsuperZZZX 
Thanks! The problem is solved now and this issue is closed.

closed 2020-03-10 by ArlenCHEN