HKUST-Aerial-Robotics/Teach-Repeat-Replan #33: take_off and landing

gouyiyong, 2020-04-26:
Hi,

Thanks for your amazing work, and I have some details I can´t understand well.
1. In the readme file section 8.4 ¨Start the quadrotor with a remote control while keeping the upper left switch be selected in A and the lower right "H" switch in the position up. At an altitude of about one meter, move the upper left switch from A to P, the quadrotor will enter a stable hovering state, and then move the lower right "H" switch down. If nothing goes wrong, the quadrotor will automatically start to execute the desired trajectory.¨  I would like to know how to take off stably. I am not using DJI remote controller, just with AT9S. In MODE A, actually during that moment no position control and position information input into the N3 autopilot and how the drone can take off stably without any position information.
2. what should I do during landing? in MODE A or P?  could you describe the landing process?

kind regards,
Yiyong

USTfgaoaa, 2020-04-27:
@gouyiyong It doesn't matter what remote controller you use. Stable means just take off softly and slowly, no special operations required. When landing, whatever mode is ok if you can safely land it on the ground, I don't think that should be a problem while playing with the dron.

gouyiyong, 2020-04-27:
> @gouyiyong It doesn't matter what remote controller you use. Stable means just take off softly and slowly, no special operations required. When landing, whatever mode is ok if you can safely land it on the ground, I don't think that should be a problem while playing with the drone.

@USTfgaoaa thanks a lot

closed 2020-04-27 by gouyiyong