HKUST-Aerial-Robotics/Teach-Repeat-Replan #34: A minor fix in pluginlib for those using Ubuntu 18.04 and ROS melodic

syangav, 2020-05-30:
In Ubuntu 18.04 and ROS melodic, the following compilation error will occur (ubuntu 16.04 & ROS kinetic works fine however):
error: expected constructor, destructor, or type conversion before ‘(’ token
 PLUGINLIB_DECLARE_CLASS(router, RouterNode, RouterNode, nodelet::Nodelet);

The solution is to change the line from: 
PLUGINLIB_DECLARE_CLASS(router, RouterNode, RouterNode, nodelet::Nodelet);
to: 
PLUGINLIB_EXPORT_CLASS(RouterNode, nodelet::Nodelet);

Note that this issue is caused by different versions of pluginlib and this similar issue will happen 3 times (CascadeControllerNode, MissionTemplateNode, RouterNode). Just using similar fixes will be fine. 

One can find the references from [1](https://github.com/KumarRobotics/msckf_vio/issues/69), [2](https://github.com/KumarRobotics/msckf_vio/pull/72), [3](https://answers.ros.org/question/323682/imu_filter_madgwick-does-not-compile-properly-on-ubuntu1804-melodic/).

bigsuperZZZX, 2020-05-31:
Thank you very much for correction. I will post it to _readme.md_, hoping to help more people that use Ubuntu 18.04 and ROS melodic.

closed 2020-05-31 by bigsuperZZZX

958117216, 2020-07-15:
太感谢了！！！！！！！！

Akashleena, 2020-12-27:
Having the same problem in ROS Noetic and Ubuntu 20.04