HKUST-Aerial-Robotics/Teach-Repeat-Replan #36: How to get APPID and ENCKEY

G1246830895, 2020-06-24:
When I follow the step-by-step tutorial to install the software, I don’t know how to check my APPID and ENCKEY. What should I do to solve this problem?

bigsuperZZZX, 2020-06-29:
Have you compiled your code with folder "[utils](https://github.com/HKUST-Aerial-Robotics/Teach-Repeat-Replan/tree/master/utils)"?
Package "uav_utils" lies in "[utils](https://github.com/HKUST-Aerial-Robotics/Teach-Repeat-Replan/tree/master/utils)" folder, which contains many packages needed by the whole system.
Remember to copy this folder to both the onboard computer and the ground station.

G1246830895, 2020-06-29:
I downloaded the source code directly from github, and then copied the "ground station" folder to the "src" folder of catkin_ws. I run catkin_make and I get the above error. What should I do to solve this problem?

bigsuperZZZX, 2020-06-29:
Just copy the "utils" folder to the same directory of "ground station" and run "catkin_make -j1".
Like this:
[image]

G1246830895, 2020-06-29:
I will try now

G1246830895, 2020-06-29:
ground_station/joy_ctrl/CMakeLists.txt:6 (find_package)

-- Could not find the required component 'uav_utils'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "uav_utils" with
  any of the following names:

    uav_utilsConfig.cmake
    uav_utils-config.cmake

  Add the installation prefix of "uav_utils" to CMAKE_PREFIX_PATH or set
  "uav_utils_DIR" to a directory containing one of the above files.  If
  "uav_utils" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  ground_station/joy_ctrl/CMakeLists.txt:6 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/gzh/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/gzh/catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed
it don't work.

G1246830895, 2020-06-29:
I can't directly copy "utils" to src, otherwise there will be a file with the same name, which can't execute catkin_make. I copied the "utils" file in "ground station" to src and deleted the file in "ground station". I run catkin_make -j1 and the error still exists.

G1246830895, 2020-06-29:
I also manipulated every folder like this, but it still doesn't work

bigsuperZZZX, 2020-06-29:
Try to place the folders like my workspace.
[image]
[image]

And before re-building, delete the "devel/" and "build/" folders.

G1246830895, 2020-06-29:
i'll try like this.

G1246830895, 2020-06-29:
Should the other folders be in files?

G1246830895, 2020-06-29:
Should the other folders be placed in the "files" folder?

bigsuperZZZX, 2020-06-29:
Don't need to.
"files" folder is the folder in [github repo](https://github.com/HKUST-Aerial-Robotics/Teach-Repeat-Replan/tree/master/files) and only some pictures in it. But it was needed by "key2joy.py", so don't delete it.

G1246830895, 2020-06-29:
Did you delete the problematic folders? When I keep "joy_ctrl", the compilation still gives an error. This error is consistent with the error when I first compiled "utils" when compiling "Ground Station"

G1246830895, 2020-06-29:
Will this cause problems for subsequent incomplete functions?

bigsuperZZZX, 2020-06-29:
Yes, it will cause problems.
Which branch are you use, master or experiment? If you use master branch, No need to change any folders or files, just put all the foldes in to a "src" folder and  type "catkin_make -j1" will be totally ok. Of course the needed solvers and other softwares shouled be installed properly in advance. The steps are listed in readme.md in detial.
And which ros version are you use? We only tested the code in kinetic but with various platforms.
Maybe re-create a blank workspace and put the files there and re-build the workspace will be ok? It sounds like you have already  modified some folders.

G1246830895, 2020-06-29:
I use the experimental part. I just created a blank workspace and then followed the steps of readme. I encountered these problems when I compiled the code of the ground station. Is there a better solution? We will conduct a real machine test at that time

G1246830895, 2020-06-29:
I use ubuntu16.04 OS and install ros-kinetic.

bigsuperZZZX, 2020-06-29:
I checked the code and found that a bug which causes the problem.
This happened due to git commands ignore blank folders which is required by CMake.
Now the bug has been fixed, you can download the repo again and it should works ok.

G1246830895, 2020-06-29:
I downloaded the code again.When I compile according to the following operation method, the problem still exists.
[image]
When I operate as I did, the result is as follows
[image]
[image]

G1246830895, 2020-06-29:
I deleted the "joy_ctrl" folder and the compilation was successful.

bigsuperZZZX, 2020-06-29:
I just found that I missed another error in that CMakelists.txt file and I have fixed it I think.
If the "joy_ctrl" package gets messing, you will not be able to perform "manual teach". If you do not need that function, deleting is ok.

G1246830895, 2020-06-29:
I think I need this function. Can this error be fixed? If this error cannot be fixed, how can I implement the teaching function?

bigsuperZZZX, 2020-06-29:
I have already fixed the bug and have just successfully compiled the latest version. Try it, and remember to use "catkin_make -j1".

G1246830895, 2020-06-30:
how did you do it? I just re-downloaded the compilation and the error still exists. Can you elaborate on your operation process?

G1246830895, 2020-06-30:
When I deleted the "joy_ctrl" folder, the compilation was successful again.

bigsuperZZZX, 2020-06-30:
Is your "ground_station/joy_ctrl/CMakeLists.txt" look like this?
[image]

The steps are:
1. Copying the files in "ground_station/" to a ros workspace src/ folder. 
2. Execute "catkin_make -j1".

G1246830895, 2020-06-30:
[image]
My steps are as follows:
1.Copying the files in "ground_station/" to a ros workspace src/ folder.
2.Execute "catkin_make -j1".
Is it possible for this reason: when I run install_tools.sh,
sudo update-alternatives --install -y /usr/bin/gcc gcc /usr/bin/gcc-5 60 --slave /usr/bin/g++ g++ /usr/bin/g++-5 &
sudo update-alternatives --install -y /usr/bin/gcc gcc /usr/bin/gcc-7 50 --slave /usr/bin/g++ g++ /usr/bin/g++-7
These two instructions will prompt me that the priority must be an integer

Hyuhao, 2020-06-30:
I think I have fixed this problem. You need to add the following commands in the package.xml file under the joy_ctrl folder

- <build_depend>uav_utils </build_depend> 

- <run_depend>uav_utils </run_depend>

G1246830895, 2020-07-01:
This problem has been solved, thank you very much

closed 2020-07-03 by USTfgaoaa