HKUST-Aerial-Robotics/Teach-Repeat-Replan #37: Onboard SDK compilation issues

G1246830895, 2020-06-30:
When I copied the code in the "Onboard_computer" folder to the ROS workspace for compilation according to the readme file, the following problems occurred:
CMake Error at onboard_computer/controller/djiros/CMakeLists.txt:36 (find_package):
  By not providing "FindDJIOSDK.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "DJIOSDK", but
  CMake did not find one.

  Could not find a package configuration file provided by "DJIOSDK" with any
  of the following names:

    DJIOSDKConfig.cmake
    djiosdk-config.cmake

  Add the installation prefix of "DJIOSDK" to CMAKE_PREFIX_PATH or set
  "DJIOSDK_DIR" to a directory containing one of the above files.  If
  "DJIOSDK" provides a separate development package or SDK, be sure it has
  been installed.
This problem is very similar to the Cmake problem caused by compiling "joy_ctrl". How should I solve it?

bigsuperZZZX, 2020-06-30:
Make sure this step is performed correctly.
[image]

G1246830895, 2020-06-30:
I did this. I followed the "step by step" tutorial for installation. The onboard computer I use is a dji manifold2 g. My Onboard-SDK is installed under "home", djiros I git to the src folder in the ROS workspace. When I compile, the same problem occurs.

bigsuperZZZX, 2020-06-30:
I have no idea about the solution. Because we compiled djiros successfully on manifold-2C.
Make sure you have properly installed Onboard-SDK via "sudo make install" and can be find from a simple CMakeLists.txt.
Maybe you can try to follow the steps from dji-developer websites to check your installation.

closed 2020-07-03 by USTfgaoaa