HKUST-Aerial-Robotics/Teach-Repeat-Replan #38: djiros compilation problem

G1246830895, 2020-07-07:
I installed the Onboard-SDK according to the tutorial and cloned djiros in the ROS workspace. When I execute the catkin_make command, it gives me the following error:
CMake Error at djiros/CMakeLists.txt:36 (find_package):
  By not providing "FindDJIOSDK.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "DJIOSDK", but
  CMake did not find one.

  Could not find a package configuration file provided by "DJIOSDK" with any
  of the following names:

    DJIOSDKConfig.cmake
    djiosdk-config.cmake

  Add the installation prefix of "DJIOSDK" to CMAKE_PREFIX_PATH or set
  "DJIOSDK_DIR" to a directory containing one of the above files.  If
  "DJIOSDK" provides a separate development package or SDK, be sure it has
  been installed.

-- Configuring incomplete, errors occurred!
See also "/home/dji/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/dji/catkin_ws/build/CMakeFiles/CMakeError.log".
Then I added "set(DJIOSDK_DIR /home/dji/Onboard-SDK/osdk-core/advanced-sensing/ori-osdk-core/cmake-modules)" to the CMakeList.txt file and recompiled, the results are as follows:
CMake Error at /home/dji/Onboard-SDK/osdk-core/advanced-sensing/ori-osdk-core/cmake-modules/DJIOSDKConfig.cmake:11 (include):
  include could not find load file:

    /home/dji/Onboard-SDK/osdk-core/advanced-sensing/ori-osdk-core/cmake-modules/djiosdkTargets.cmake
Call Stack (most recent call first):
  djiros/CMakeLists.txt:36 (find_package)

CMake Warning (dev) at /home/dji/Onboard-SDK/osdk-core/advanced-sensing/ori-osdk-core/cmake-modules/DJIOSDKConfig.cmake:14 (get_target_property):
  Policy CMP0045 is not set: Error on non-existent target in
  get_target_property.  Run "cmake --help-policy CMP0045" for policy details.
  Use the cmake_policy command to set the policy and suppress this warning.

  get_target_property() called with non-existent target "djiosdk-core".
Call Stack (most recent call first):
  djiros/CMakeLists.txt:36 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- djiros: 7 messages, 21 services
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
DJIOSDK_DEPENDENCIES
    linked by target "djiros_node" in directory /home/dji/catkin_ws/src/djiros

-- Configuring incomplete, errors occurred!
See also "/home/dji/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/dji/catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed
How can i solve this problem?

bigsuperZZZX, 2020-07-09:
Our djiros is modified from the official Onboard-SDK-ROS, so it needs OSDK like the official dji ROS package does.
Therefore, as I mentioned in [issue#37](https://github.com/HKUST-Aerial-Robotics/Teach-Repeat-Replan/issues/37), you have to make sure the official Onboard-SDK-ROS can run successfully. Then our djiros can be compiled as will.
Since the platform you use is different from ours, we are unable to solve these compiling errors.

G1246830895, 2020-07-09:
We are currently using a folder 2-c, and I made sure that I installed the Onboard-SDK correctly. However, I still can't solve this problem.

G1246830895, 2020-07-09:
I successfully solved the last problem. But after I copied the code in onboard_computer to the newly created ros workspace and compiled it, the following problems occurred:
/home/dji/catkin_ws/src/controller/djiros/src/djiros/DjiRosAuthority.cpp: In static member function ‘static void DjiRos::on_authority_ack(DJI::OSDK::Vehicle*, DJI::OSDK::RecvContainer, DJI::OSDK::UserData)’:
/home/dji/catkin_ws/src/controller/djiros/src/djiros/DjiRosAuthority.cpp:32:32: error: ‘PackageMin’ is not a member of ‘DJI::OSDK::OpenProtocol’
   if (recvFrame.recvInfo.len - OpenProtocol::PackageMin <= (int) sizeof(uint16_
                                ^
controller/djiros/CMakeFiles/djiros_node.dir/build.make:137: recipe for target 'controller/djiros/CMakeFiles/djiros_node.dir/src/djiros/DjiRosAuthority.cpp.o' failed
make[2]: *** [controller/djiros/CMakeFiles/djiros_node.dir/src/djiros/DjiRosAuthority.cpp.o] Error 1
CMakeFiles/Makefile2:1832: recipe for target 'controller/djiros/CMakeFiles/djiros_node.dir/all' failed
make[1]: *** [controller/djiros/CMakeFiles/djiros_node.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j1" failed
how can i solve it?

G1246830895, 2020-07-10:
We have now compiled all the code. As for the previous problem, we changed the library name you provided and solved the problem. What external network card do you choose? We want to use the same network card as you.

USTfgaoaa, 2020-07-10:
@G1246830895 I don't think the wireless card warrants a discussion here under this research-oriented project. You can choose whatever wireless card with nothing different I guess.

closed 2020-07-10 by USTfgaoaa