HKUST-Aerial-Robotics/Teach-Repeat-Replan #39: SpatialTemporalTrajectory

hesuieins, 2020-07-07:
Hi,

Thanks for the amazing open source work. I have some details that I can't understand totally.
The /tr_node/space_time_traj topic seems to store the information of the desired path generated by the /tr_node. And I'm confusing, what's the meaning of the polynomial coefficients of the trajectory in the SpatialTemporalTrajectory. How can I rebuild the trajectory by using these topic?
Thanks for your reply in advance!

Best wishes,
Suiyi

bigsuperZZZX, 2020-07-09:
"space_time_traj" is a Bézier curve the details can be viewed from our paper "Teach-Repeat-Replan: A Complete and Robust System for Aggressive Flight in Complex Environments".
If you just want to rebuild this trajectory, refer to the the [traj_server.cpp](https://github.com/HKUST-Aerial-Robotics/Teach-Repeat-Replan/blob/master/local_replanner/local_replanner/src/traj_server.cpp), which receives this "space_time_traj" and converts it to drone control commands in function "rcvTrajectoryCallBack".

USTfgaoaa, 2020-07-10:
@hesuieins Please check the above comment and close this issue If the problem solved.

hesuieins, 2020-07-10:
Thanks for your reply！

closed 2020-07-10 by hesuieins