HKUST-Aerial-Robotics/Teach-Repeat-Replan #43: Some problem about CUDA

LTianyyi, 2020-07-27:
Thanks for you and your team brilliant work! However, I faced with some questions about the GPU.

I had run `./trr_simulation.sh` successfully  without GPU. When I set the `set(ENABLE_CUDA true)` and change `CUDA_NVCC_FLAGS` in both `polyhedron_generator` and `local_sensing`, I run `catkin_make` successfully but there are some errors (I use RTX2060 and CUDA10.0).

I find that when I set using CUDA in `local_sensing`, there are no problems. But when I set using CUDA in  `polyhedron_generator`, when I finish teaching, it has error at mosek.
The error shows as follow:

```
[polyhedron_generator]{GPU} finish gpu cluster
start
[ WARN] [1595505451.593957565]: [trr_global_planner] Enter in autonomous mode
[ WARN] [1595505451.594048104]: [trr_global_planner] Add extra polytopes to connecting path
[ WARN] [1595505451.594147658]: [trr_global_planner] Start coordinate descent-based spatial-temporal joint optimization
[ WARN] [1595505451.605865117]: [trr_global_planner] Time consumation of spatial optimization is: 0.011662
Stack trace (most recent call last):
#5    Object "/home/tianyili/catkin_ws/src/Teach-Repeat-Replan/global_planner/third_party/mosek/lib/mosek8_1/libmosek64.so.8.1", at 0x7fc721e9175e, in 
#4    Object "/usr/lib/libtcmalloc_minimal.so.4", at 0x7fc724076cda, in __libc_calloc
#3    Object "/usr/lib/libtcmalloc_minimal.so.4", at 0x7fc72406dbe9, in tcmalloc::ThreadCache::FetchFromCentralCache(unsigned long, unsigned long)
#2    Object "/usr/lib/libtcmalloc_minimal.so.4", at 0x7fc72406ac0e, in tcmalloc::CentralFreeList::RemoveRange(void**, void**, int)
#1    Object "/usr/lib/libtcmalloc_minimal.so.4", at 0x7fc72406ab59, in tcmalloc::CentralFreeList::FetchFromOneSpansSafe(int, void**, void**)
#0    Object "/usr/lib/libtcmalloc_minimal.so.4", at 0x7fc72406a883, in tcmalloc::CentralFreeList::FetchFromOneSpans(int, void**, void**)
Segmentation fault (Invalid permissions for mapped object [0x7fc6d7800000])
start
================================================================================REQUIRED process [tr_node-1] has died!
process has died [pid 9056, exit code -11, cmd /home/tianyili/catkin_ws/devel/lib/trr_global_planner/tr_node ~waypoints:=/waypoint_generator/waypoints ~odometry:=/drone_1/visual_slam/odom ~map:=/random_complex/global_map ~joystick:=/joy __name:=tr_node __log:=/home/tianyili/.ros/log/97fdebf8-ccdb-11ea-be58-c0b883013038/tr_node-1.log].
log file: /home/tianyili/.ros/log/97fdebf8-ccdb-11ea-be58-c0b883013038/tr_node-1*.log
Initiating shutdown!
================================================================================
[pcl_render_node-2] killing on exit
[tr_node-1] killing on exit
terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::lock_error> >'
  what():  boost: mutex lock failed in pthread_mutex_lock: Invalid argument
shutting down processing monitor...
... shutting down processing monitor complete
done
```

Could you kindly give me some hints? Thanks!

USTfgaoaa, 2020-07-27:
@LTianyyi Mosek should not be affected by whether or not to use CUDA. 
Can you check if the corridor returned by the polyhedron_generator looks good? I mean check if the corridor has no broken part, and all trajectory segment has normal time durations, to make sure while calling Mosek it's not solving an ill-posed problem.

LTianyyi, 2020-07-27:
Thanks for your help, but I had checked and I found that it might not the problem of Mosek. I used keyboard to control the UAV, and when I pressed the `N` or `M` to clean or repeat, the program would error and close `tr_node`, the terminal  shows as follow. 
I'm not sure what's the problem, that problem did not arise when I closed CUDA.

Thats' the error after pressing `N`:
```
[ WARN] [1595833120.090726687]: [trr_global_planner] Enter in maunal flight mode
bezier curves visualization num: 0
[polyhedron_generator]{GPU} finish gpu cluster
Stack trace (most recent call last):
#22   Object "[0xffffffffffffffff]", at 0xffffffffffffffff, in 
#21   Object "/home/tianyili/catkin_ws/devel/lib/trr_global_planner/tr_node", at 0x42bea8, in _start
#20   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f8687af183f, in __libc_start_main
#19   Object "/home/tianyili/catkin_ws/devel/lib/trr_global_planner/tr_node", at 0x42afdc, in main
#18   Object "/opt/ros/kinetic/lib/libroscpp.so", at 0x7f868b760ef4, in ros::spinOnce()
#17   Object "/opt/ros/kinetic/lib/libroscpp.so", at 0x7f868b71f23a, in ros::CallbackQueue::callAvailable(ros::WallDuration)
#16   Object "/opt/ros/kinetic/lib/libroscpp.so", at 0x7f868b71d837, in ros::CallbackQueue::callOneCB(ros::CallbackQueue::TLS*)
#15   Object "/opt/ros/kinetic/lib/libroscpp.so", at 0x7f868b777d8c, in ros::SubscriptionQueue::call()
#14   Object "/home/tianyili/catkin_ws/devel/lib/trr_global_planner/tr_node", at 0x46060a, in ros::SubscriptionCallbackHelperT<nav_msgs::Odometry_<std::allocator<void> >, void>::call(ros::SubscriptionCallbackHelperCallParams&)
#13   Object "/home/tianyili/catkin_ws/devel/lib/trr_global_planner/tr_node", at 0x44a6d3, in boost::detail::function::void_function_invoker1<void (*)(nav_msgs::Odometry_<std::allocator<void> >), void, nav_msgs::Odometry_<std::allocator<void> > >::invoke(boost::detail::function::function_buffer&, nav_msgs::Odometry_<std::allocator<void> >)
#12   Object "/home/tianyili/catkin_ws/devel/lib/trr_global_planner/tr_node", at 0x442c28, in rcvOdometryCallBack(nav_msgs::Odometry_<std::allocator<void> >)
#11   Object "/home/tianyili/catkin_ws/devel/lib/trr_global_planner/tr_node", at 0x498003, in polyhedronGenerator::corridorIncreGeneration(std::vector<Eigen::Matrix<double, 3, 1, 0, 3, 1>, std::allocator<Eigen::Matrix<double, 3, 1, 0, 3, 1> > > const&, decomp_ros_msgs::PolyhedronArray_<std::allocator<void> >&)
#10   Object "/home/tianyili/catkin_ws/devel/lib/trr_global_planner/tr_node", at 0x440d55, in Eigen::Polyhedron::setVertices(Eigen::Matrix<double, -1, -1, 0, -1, -1> const&)
#9    Object "/home/tianyili/catkin_ws/devel/lib/trr_global_planner/tr_node", at 0x440a02, in Eigen::Polyhedron::setVrep(Eigen::Matrix<double, -1, -1, 0, -1, -1> const&, Eigen::Matrix<double, -1, 1, 0, -1, 1> const&)
#8    Object "/home/tianyili/catkin_ws/devel/lib/trr_global_planner/tr_node", at 0x44088e, in Eigen::Polyhedron::hvrep(Eigen::Matrix<double, -1, -1, 0, -1, -1> const&, Eigen::Matrix<double, -1, 1, 0, -1, 1> const&, bool)
#7    Object "/home/tianyili/catkin_ws/devel/lib/trr_global_planner/tr_node", at 0x43fa67, in Eigen::Polyhedron::doubleDescription(Eigen::Matrix<double, -1, -1, 0, -1, -1> const&, bool)
#6    Object "/usr/lib/x86_64-linux-gnu/libcdd.so.0", at 0x7f86888b59e9, in dd_DDMatrix2Poly
#5    Object "/usr/lib/x86_64-linux-gnu/libcdd.so.0", at 0x7f86888b541d, in dd_CreatePolyhedraData
#4    Object "/usr/lib/libtcmalloc_minimal.so.4", at 0x7f868b9e9dca, in __libc_malloc
#3    Object "/usr/lib/libtcmalloc_minimal.so.4", at 0x7f868b9e3be9, in tcmalloc::ThreadCache::FetchFromCentralCache(unsigned long, unsigned long)
#2    Object "/usr/lib/libtcmalloc_minimal.so.4", at 0x7f868b9e0c0e, in tcmalloc::CentralFreeList::RemoveRange(void**, void**, int)
#1    Object "/usr/lib/libtcmalloc_minimal.so.4", at 0x7f868b9e0b59, in tcmalloc::CentralFreeList::FetchFromOneSpansSafe(int, void**, void**)
#0    Object "/usr/lib/libtcmalloc_minimal.so.4", at 0x7f868b9e0883, in tcmalloc::CentralFreeList::FetchFromOneSpans(int, void**, void**)
Segmentation fault (Invalid permissions for mapped object [0x7f863f800000])
================================================================================REQUIRED process [tr_node-1] has died!
process has died [pid 17480, exit code -11, cmd /home/tianyili/catkin_ws/devel/lib/trr_global_planner/tr_node ~waypoints:=/waypoint_generator/waypoints ~odometry:=/drone_1/visual_slam/odom ~map:=/random_complex/global_map ~joystick:=/joy __name:=tr_node __log:=/home/tianyili/.ros/log/5b57f16e-cfd6-11ea-97a0-c0b883013038/tr_node-1.log].
log file: /home/tianyili/.ros/log/5b57f16e-cfd6-11ea-97a0-c0b883013038/tr_node-1*.log
Initiating shutdown!
================================================================================
[pcl_render_node-2] killing on exit
```

And that's the error after pressing `M`:

```
start
[ WARN] [1595833563.378537205]: [trr_global_planner] Enter in autonomous mode
[ WARN] [1595833563.378711910]: [trr_global_planner] Add extra polytopes to connecting path
[ WARN] [1595833563.378844723]: [trr_global_planner] Start coordinate descent-based spatial-temporal joint optimization
[ WARN] [1595833563.384768106]: [trr_global_planner] Time consumation of spatial optimization is: 0.005853
Stack trace (most recent call last):
#4    Object "/home/tianyili/catkin_ws/src/Teach-Repeat-Replan/global_planner/third_party/mosek/lib/mosek8_1/libmosek64.so.8.1", at 0x7f68710ef75e, in 
[... cut ...]
process has died [pid 22260, exit code -11, cmd /home/tianyili/catkin_ws/devel/lib/trr_global_planner/tr_node ~waypoints:=/waypoint_generator/waypoints ~odometry:=/drone_1/visual_slam/odom ~map:=/random_complex/global_map ~joystick:=/joy __name:=tr_node __log:=/home/tianyili/.ros/log/76bbf2ec-cfd7-11ea-97a0-c0b883013038/tr_node-1.log].
log file: /home/tianyili/.ros/log/76bbf2ec-cfd7-11ea-97a0-c0b883013038/tr_node-1*.log
Initiating shutdown!
================================================================================
[pcl_render_node-2] killing on exit
```

Could you please give me some guidance? Thanks!

USTfgaoaa, 2020-07-27:
@LTianyyi Yes it does look like CUDA issues. Maybe the memory leak? I pre-allocated CUDA memory in the polygon_generator, so you may try to modify the allocated memory size. Or you may try different map resolution (It decides polygon resolution) and different polygon inflation iteration numbers in the launch file.

LTianyyi, 2020-07-29:
I changed the resolution of the map, but it did not work. I will try to learn CUDA and try to change the memory. And there is another problem. Sometimes it will have a collision with the obstacle, but the map is statics without dynamical obstacles. Is that the problem with replanning because it cannot make sure of the feasible solution?
The video recorded is as follows, thanks!
[test_repeat.zip](https://github.com/HKUST-Aerial-Robotics/Teach-Repeat-Replan/files/4993140/test_repeat.zip)

USTfgaoaa, 2020-07-29:
@LTianyyi Firstly, the basic claim is, here the global planning is always strictly feasible (safe). 
So, in a static map, the only possibility to detect a collision is the noise in the depth observation. In your case, since the drone's sensor has a limited sensing horizon, when it detects some noise (It can't judge whether this is noise or true obstacle), it may not have enough time to generate a feasible replan.
To solve this issue, you may set a higher safety margin in global planning to alleviate the effect of noise, or improve the success rate of the local planner. Since we use no initialization in the local planning, the local planner in this repo can't be as robust as a complete planner. For example, you may search a safe path with A* and use it to initialize the local trajectory optimization. It will help a lot.

LTianyyi, 2020-07-29:
Thanks for your solution!

closed 2020-07-29 by LTianyyi