HKUST-Aerial-Robotics/Teach-Repeat-Replan #44: Path planning in actual flight

G1246830895, 2020-07-27:
We successfully planned an executable path, but when we generate the optimized path with teaching_planning.sh, the optimized path always starts from the origin of the coordinate system. We use the remote control to control the aircraft correctly, but the aircraft does not execute the path we planned and prints "[ INFO] [1595826566.979165352]: Integration saturate for axis 1, value=0.400". We watched your operation video and found that your drone was in flight during the teach phase, and our situation would not happen. How did you do it?

G1246830895, 2020-07-27:
[image]
[image]

USTfgaoaa, 2020-07-27:
@G1246830895 Of course you shoudl launch the drone and trigger the trajectory generation. There is no auto-take off functionality.

G1246830895, 2020-07-27:
Yes, we did. The path we plan always starts from the origin of the coordinates, not in the air. So after we take off the drone, the drone will not execute the planned path

USTfgaoaa, 2020-07-27:
@G1246830895 The traj generated "always must" starts from the current state of the drone, not any coordinate no matter the origin or the air. Please make sure you correctly set all params/topics and figure out the system msgs flow.

G1246830895, 2020-07-27:
[image]
The path we planned is like this

closed 2020-08-18 by G1246830895