HKUST-Aerial-Robotics/Teach-Repeat-Replan #46: Cannot find a safe solution

itsalpha, 2020-11-11:
Hello, I have always liked your work very much, but when I went to retest, the following questions appeared, I would like to ask you to help answer them.
1.
[image]
2.
[image]
Thank you very much!

itsalpha, 2020-11-11:
I didn't have flight control and camera drive on the ground station.will it matter?

bigsuperZZZX, 2020-11-12:
I have no idea about the exact problem with the given figures.
The first figure is some plugin errors related to rviz, which does not matter.
I suggest you start testing from a simple teaching trajectory that contains as fewer polyhedrons as possible, then check that there is no disconnection(break) between adjacent polyhedrons from rviz.
In addition, make sure that the drone is inside one polyhedron once you press the start button.
Flight control and camera drive will not affect the global planner.

itsalpha, 2020-11-12:
Thank you very much for your reply. I am testing the model-balcony I brought in the project now, but I can not solve an optimized path by modifying the resolution and map size. There is also in the global-planner-exp.launch file trr-global-planner node file-path need to be set?

itsalpha, 2020-11-12:
[image]

bigsuperZZZX, 2020-11-12:
Try a teaching path like this
[image]

itsalpha, 2020-11-12:
@bigsuperZZZX 
[image]
not ok.

USTfgaoaa, 2020-11-12:
@itsalpha pilot the drone to go around an obstacle and back. Try it again.

itsalpha, 2020-11-12:
@USTfgaoaa sorry,my dear teacher,it still doesn't work.But it can be run in simulation.
[image]

USTfgaoaa, 2020-11-12:
@itsalpha So you are going to conduct real-world experiments? Make sure you know how this system functions. For example, for real flight here the odometry name in the launch file should be the topic name of your onboard estimated pose.

itsalpha, 2020-11-12:
@bigsuperZZZX I'm sorry to brother you again,after i modified the launch file,i will see the generated track ,but the effect id not good,is this the reason why the launch file is not configured correctly?
[image]
1.gobal_planner.luanch
```c
<launch>

<arg name="map_size_x" default="20.0"/>
<arg name="map_size_y" default="20.0"/>
<arg name="map_size_z" default="5.0"/>

<arg name="init_x" default="0.0"/>
<arg name="init_y" default="0.0"/>
<arg name="init_z" default="1.0"/>

<arg name="max_vel" default="0.4"/>
<arg name="max_acc" default="0.8"/>

<arg name="resolution" default="0.33"/>
<arg name="poly_order" default="5"/>

  <node pkg="trr_global_planner" type="tr_node" name="tr_node" output="screen" required = "true">
      <remap from="~odometry"    to="/drone_1/visual_slam/odom"    />
      <remap from="~map"         to="/cloud_pcd"   /> 
      <remap from="~joystick"    to="/joy"/> 
      <remap from="~trigger"     to="/time_optimal_traj_server/start_pose"/>

      <param name="write_path"  value="false" /> 
      <param name="read_path"   value="false" /> 

      <param name="optimization/poly_order"  value="$(arg poly_order)"     /> 
      <param name="optimization/min_order"   value="3.0"   /> 
      
      <param name="map/x_size"       value="$(arg map_size_x)"/>
      <param name="map/y_size"       value="$(arg map_size_y)"/>
      <param name="map/z_size"       value="$(arg map_size_z)"/>
      
      <param name="map/map_margin"   value="0.1" />
      <param name="map/resolution"   value="$(arg resolution)" />
      
      <param name="planning/rho_time"      value="10.0"/>
      <param name="planning/max_vel"       value="$(arg max_vel)" />
      <param name="planning/max_acc"       value="$(arg max_acc)" />
      <param name="planning/max_d_acc"     value="1.0" />
      <param name="planning/max_inf_iter"  value="10"  />
      <param name="planning/max_clu_iter"  value="5"  />

      <param name="z_cost_weight"  value="100.0"  />

      <param name="vis/vis_traj_width"     value="0.1  "/>
      <param name="your_file_path"         value="/home/zx/TRR_ws/catkin_ws/path_pts"/>
  </node>
</launch>
```
2.joy_ctrl.launch
```c
<?xml version="1.0"?>

<launch>
  <node pkg="joy_ctrl" type="joy_ctrl" name="joy_ctrl_sim" output="screen">
    <remap from="~joy" to="/joy"/>
    <remap from="~uav_odom_sim" to="/drone_1/visual_slam/odom"/>
    <param name="start_x"   type="double" value=" 0.0"/>
    <param name="start_y"   type="double" value=" 0.0"/>
    <param name="start_z"   type="double" value=" 1.0"/>
    <param name="start_yaw" type="double" value="0.0"/>

    <param name="min_z"   type="double" value=" 0.0"/>
    <param name="max_z"   type="double" value=" 2.0"/>

    <param name="hori_scale" type="double" value="0.03"/>
[... 125 lines cut ...]
/tf
/tf_static
/time_optimal_traj_server/check_pts
/time_optimal_traj_server/current_pose
/time_optimal_traj_server/desired_acceleration
/time_optimal_traj_server/desired_position
/time_optimal_traj_server/desired_velocity
/time_optimal_traj_server/start_pose
/time_optimal_traj_server/vis_yaw_cmd
/tr_node/space_time_traj
/traj_start_trigger
/view
/vins_estimator/camera_pose
/vins_estimator/camera_pose_visual
/vins_estimator/extrinsic
/vins_estimator/imu_propagate
/vins_estimator/key_poses
/vins_estimator/keyframe_point
/vins_estimator/keyframe_pose
/vins_estimator/margin_cloud
/vins_estimator/odometry
/vins_estimator/path
/vins_estimator/point_cloud

```

closed 2020-11-14 by itsalpha

xyft123, 2021-05-22:
> @bigsuperZZZX I'm sorry to brother you again,after i modified the launch file,i will see the generated track ,but the effect id not good,is this the reason why the launch file is not configured correctly?
> [image]
> 1.gobal_planner.luanch
> 
> ```c
> <launch>
> 
> <arg name="map_size_x" default="20.0"/>
> <arg name="map_size_y" default="20.0"/>
> <arg name="map_size_z" default="5.0"/>
> 
> <arg name="init_x" default="0.0"/>
> <arg name="init_y" default="0.0"/>
> <arg name="init_z" default="1.0"/>
> 
> <arg name="max_vel" default="0.4"/>
> <arg name="max_acc" default="0.8"/>
> 
> <arg name="resolution" default="0.33"/>
> <arg name="poly_order" default="5"/>
> 
>   <node pkg="trr_global_planner" type="tr_node" name="tr_node" output="screen" required = "true">
>       <remap from="~odometry"    to="/drone_1/visual_slam/odom"    />
>       <remap from="~map"         to="/cloud_pcd"   /> 
>       <remap from="~joystick"    to="/joy"/> 
>       <remap from="~trigger"     to="/time_optimal_traj_server/start_pose"/>
> 
>       <param name="write_path"  value="false" /> 
>       <param name="read_path"   value="false" /> 
> 
>       <param name="optimization/poly_order"  value="$(arg poly_order)"     /> 
>       <param name="optimization/min_order"   value="3.0"   /> 
>       
>       <param name="map/x_size"       value="$(arg map_size_x)"/>
>       <param name="map/y_size"       value="$(arg map_size_y)"/>
>       <param name="map/z_size"       value="$(arg map_size_z)"/>
>       
>       <param name="map/map_margin"   value="0.1" />
>       <param name="map/resolution"   value="$(arg resolution)" />
>       
>       <param name="planning/rho_time"      value="10.0"/>
>       <param name="planning/max_vel"       value="$(arg max_vel)" />
>       <param name="planning/max_acc"       value="$(arg max_acc)" />
>       <param name="planning/max_d_acc"     value="1.0" />
>       <param name="planning/max_inf_iter"  value="10"  />
>       <param name="planning/max_clu_iter"  value="5"  />
> 
>       <param name="z_cost_weight"  value="100.0"  />
> 
>       <param name="vis/vis_traj_width"     value="0.1  "/>
>       <param name="your_file_path"         value="/home/zx/TRR_ws/catkin_ws/path_pts"/>
>   </node>
> </launch>
> ```
> 
> 2.joy_ctrl.launch
> 
> ```c
> <?xml version="1.0"?>
> 
> <launch>
>   <node pkg="joy_ctrl" type="joy_ctrl" name="joy_ctrl_sim" output="screen">
>     <remap from="~joy" to="/joy"/>
>     <remap from="~uav_odom_sim" to="/drone_1/visual_slam/odom"/>
>     <param name="start_x"   type="double" value=" 0.0"/>
>     <param name="start_y"   type="double" value=" 0.0"/>
>     <param name="start_z"   type="double" value=" 1.0"/>
>     <param name="start_yaw" type="double" value="0.0"/>
> 
>     <param name="min_z"   type="double" value=" 0.0"/>
[... 124 lines cut ...]
> /tf_static
> /time_optimal_traj_server/check_pts
> /time_optimal_traj_server/current_pose
> /time_optimal_traj_server/desired_acceleration
> /time_optimal_traj_server/desired_position
> /time_optimal_traj_server/desired_velocity
> /time_optimal_traj_server/start_pose
> /time_optimal_traj_server/vis_yaw_cmd
> /tr_node/space_time_traj
> /traj_start_trigger
> /view
> /vins_estimator/camera_pose
> /vins_estimator/camera_pose_visual
> /vins_estimator/extrinsic
> /vins_estimator/imu_propagate
> /vins_estimator/key_poses
> /vins_estimator/keyframe_point
> /vins_estimator/keyframe_pose
> /vins_estimator/margin_cloud
> /vins_estimator/odometry
> /vins_estimator/path
> /vins_estimator/point_cloud
> ```

I've met the same problem, how did you finally solve it?