HKUST-Aerial-Robotics/Teach-Repeat-Replan #48: 在模拟器中进行试验时遇到错误Reach percent_lower_limit 0.050000

sbdjyl, 2021-06-06:
你好，我在用这个项目中n3ctrl 实验时，出现错误代码

[WARN] [1622970831.658573214]: [n3ctrl.hover] Reach percent_lower_limit 0.050000

我先在模拟器中让飞机起飞，随后切换到A模式进入api控制，就遇到了这段代码。

模拟器中的无人接在飞行中直接坠机，请问这是遇到了什么问题？

bigsuperZZZX, 2021-06-08:
报的这个警告只会导致无人机高度不断上升，表示控制器认为当油门一路下降到5%那么小的时候飞机还不能悬停。 
这与模拟器中坠毁没有直接关系。
猜想是topic没接全，比如imu的消息漏掉了

sbdjyl, 2021-06-09:
您好，我对无人机imu和指南针进行了矫正，更换了电池之后，在模拟器中不再坠机了。
我想要用n3ctrl进行无人机的实际控制，采用fastplanner规划路径。
ctrl_md.launch文件修改为

```
<launch>
	<arg name="uavname" default="fpv" />
        <arg name="odom_topic" value="/vins_estimator/odometry" />

	<node pkg="n3ctrl" type="n3ctrl_node" name="n3ctrl" output="screen">
		<remap from="~joy" to="/djiros/rc" />
        	<remap from="~odom" to="$(arg odom_topic)" />
		<remap from="~imu" to="/djiros/imu" />
		<remap from="~cmd" to="/planning/pos_cmd" />

		<remap from="~desire_pose" to="/desire_pose" />
		<remap from="~traj_start_trigger" to="/traj_start_trigger" />
		<remap from="~ctrl" to="/djiros/ctrl" />

                <rosparam command="load" file="$(find n3ctrl)/config/ctrl_param_$(arg uavname).yaml" />
        <param name="work_mode" value="real-time" />
        <param name="js_ctrl_mode" value="feedback" />
	</node>
 
</launch>
```

但是在模拟器中，无人机并没有按照规划路径飞行，而是一直悬停在原点。
终端中收到提示：

```
[ INFO] [1623203135.428662773]: Integration saturate for axis 0, value=0.400
[ INFO] [1623203135.428689111]: Integration saturate for axis 1, value=0.400
[ INFO] [1623203135.428702514]: Integration saturate for axis 2, value=0.400
[ WARN] [1623203135.428745828]: [n3ctrl.hover] Reach percent_lower_limit 0.050000
```

请问我该如何解决这一问题？

bigsuperZZZX, 2021-06-09:
你说的模拟器是啥，是dji assistant中的模拟器吗？根据我们曾经的使用经验，这个模拟器的表现就是奇奇怪怪的，参考价值不大。
建议直接实机飞行，前期可以拆掉螺旋桨，通过听电机声音来大致估计控制链路是否正确

bigsuperZZZX, 2021-09-17:
如果没有更多问题，我就关闭这个issue了

closed 2021-09-17 by bigsuperZZZX