HKUST-Aerial-Robotics/Teach-Repeat-Replan #50: wait for imu... and the big druft

pang0224, 2021-09-18:
您好，非常感谢您项目的开源分享！
在experiment中，我在遇到了一些问题，具体情况如下：
我发现在运行VINS-Fusion中的vins_node时，总是会提示我：wait for imu
我在错误处（estimator.cpp  L166）做了一下打印

[image]

我发现在线估计的时间差td总是正值，而且td会随时间不断增大，如下所示：

USE_IMU = 1, feature.first.toSec() = 1631947477.621126
accBuf.empty = 0, td = 0.223983
t=feature.first.toSec()+td = 1631947477.845110 < accBuf.back().first = 1631947477.799380
wait for imu2 ... 
我通过kalibr离线标定的td为-0.0068，并且配置在config文件中
当我不使用时间在线估计（estimate_td=0）时，不会报wait for imu的错误，但是在./mapping的rviz中会出现严重的漂移

[image]

我关闭了D435i的IR结构光，并做了kalibr离线标定，配置了config文件，N3的串口也设定好了

我还将/djiros/imu换成了/camera/imu进行尝试，但结果似乎是一样的

查了许多资料，但也没能找到有效的解决方法，希望能够得到您的解答

除此之外我注意到output文件夹中生成的转换矩阵的数据似乎不太对，不知道会有什么影响

extrinsic_parameter.csv:

body_T_cam0: !!opencv-matrix
   rows: 4
   cols: 4
   dt: d
   data: [ -4.8894667963096716e-01, 8.8920376763920406e-01,
       4.3082156593765397e-01, 2.4204037556535291e+03,
       -9.5155803980695841e-01, -5.5933530995667868e-01,
       8.4980294589126162e-02, -5.5900832429765229e+04,
       2.6613503891721402e-01, -3.4928643432856077e-01,
       9.0596482287622315e-01, -1.0536773009529390e+05, 0., 0., 0., 1. ]
body_T_cam1: !!opencv-matrix
   rows: 4
   cols: 4
   dt: d
   data: [ 7.5045347504791482e-01, 6.5410542132483693e-02,
       6.5767852539553351e-01, 1.9753958570554408e+06,
       -5.4031969467269259e-01, -5.1235085922176560e-01,
       6.6749623564745864e-01, -3.8842305466219079e+05,
       3.8062344822319738e-01, -8.5628152965753856e-01,
       -3.4915287859336019e-01, 1.6016689124682453e+05, 0., 0., 0., 1. ]

下面时我用calibr离线标定的结果

T_ci:  (imu0 to cam0):               # body_T_cam0:
[[-0.03259658 -0.99944667  0.00661995 -0.00192054]
 [ 0.0565703  -0.00845779 -0.99836279 -0.07663109]
 [ 0.99786636 -0.03216872  0.05681469 -0.12478118]
 [ 0.          0.          0.          1.        ]]

T_ci:  (imu0 to cam1):           ## body_T_cam1:
[[-0.01973357 -0.99978779  0.00591221 -0.05371594]
 [ 0.05422067 -0.00697481 -0.99850462 -0.07627765]
 [ 0.99833396 -0.01938349  0.0543468  -0.12611655]
 [ 0.          0.          0.          1.        ]]

期待您的回复，非常感谢！

bigsuperZZZX, 2021-09-18:
根据我们的使用经验，左右相机外参矩阵中分别只有一个参数是至关重要的：就是translation的三个参数中影响左目右目间距计算的那个参数，其差值一定要严格等于相机基线，其余的旋转参数直接给1/-1/0即可，其余的translation随便拿尺子量一下就行。

td也是一个至关重要的参数，但通常应该是在  -0.01附近的，初值也通常给 -0.01即可。优化出正值来说明一定是其它参数给的过于离谱，优化发散了。

看灰度相机，注意屋里的灯是不是频闪的，频闪灯下VINS很容易挂。

这个CPU版的VINS需要较高的CPU性能，作为参考，妙算是可以的，NX或者TX2会gg

pang0224, 2021-09-18:
我确实使用的是TX2，目前妙算似乎买不到了，我先重新检查一下参数矩阵，重新尝试一下，不行的话再试一下其他的上位机，非常感谢您的回复！

bigsuperZZZX, 2021-09-18:
那就用 https://github.com/pjrambo/VINS-Fusion-gpu 这个来跑，不过这个需要比较多的配置才能稳定跑起来。

或者搜索 “VINS-Fusion-GPU”后把这些都试试吧：
[image]

pang0224, 2021-09-18:
好的，非常感谢！

pang0224, 2021-09-22:
您好，我最开始发现我的转换矩阵设定反了，根据您的配置文件，应该是从相机到imu的转换矩阵，重新设定了一下发现还是不行，问题还是出在计算出的时间差td和output文件夹中在线生成的转换矩阵上，我又将VINS-Fusion替换成gpu版的，我发现td和新的在线计算的转换矩阵和我标定得到的数据是一致的，也不会产生大的漂移了，所以最终的问题可能还是归结到计算能力，这个问题解决了，谢谢您的帮助！

但在接下来的部分我又遇到了新的问题，就是在teaching阶段中的，teach后按下auto无法生成规划的轨迹，如下图所示
[image]

[image]

我查看了前面的issue，更改了一下global_planner_exp.launch中的resolution，分别改为0.1和0.4又尝试了一下，但结果还是失败。
0.1

[image]

0.4

[image]

不知道我能否通过调试其他参数来解决这个问题，您有什么好的建议吗？

我注意到当我按下auto后，rviz中的无人机模型会自动回到原点(地面)，那一段轨迹很陡峭，是不是由于这段轨迹导致的？

期待您的答复，谢谢！

pang0224, 2021-09-22:
您好，我最开始发现我的转换矩阵设定反了，根据您的配置文件，配置的应该是从相机到imu的转换矩阵。
我重新设定了配置文件，但发现还是不行，问题还是出在计算出的时间差td和output文件夹中在线生成的转换矩阵上，我又将VINS-Fusion替换成gpu版的，我发现td和新的在线计算的转换矩阵和我标定得到的数据是一致的，也不会产生大的漂移了，所以最终的问题可能还是归结到计算能力，这个问题解决了，谢谢您的帮助！

closed 2021-09-22 by pang0224

bigsuperZZZX, 2023-11-11:
生成的output文件夹中变换矩阵是对的，可以直接拷贝进配置文件

chobitsfan, 2024-04-08:
Hi @bigsuperZZZX 您好，想請教一下td的意思是:
1. camera曝光和imu取樣的時間差
[image]
或是
2. camera timestamp和imu timestamp的時差，因為camera和imu的timestamp來自不同的timer，兩個timer間的時間差。類似北京時間和東京時間差一小時
[image]
請問應該是哪一種呢? 非常感謝您

chobitsfan, 2024-04-23:
Hi @bigsuperZZZX Thank you very much for replying