HKUST-Aerial-Robotics/Teach-Repeat-Replan #54: Invoking "make -j1" failed

shiyunyan7, 2021-12-28:
您好，非常感谢您项目的开源分享！我在进行“Build on ROS”，进行catkin_make -j1时，遇见如下错误：
In file included from /usr/include/pcl-1.10/pcl/io/file_io.h:41:0,
                 from /usr/include/pcl-1.10/pcl/io/pcd_io.h:44,
                 from /opt/ros/noetic/include/pcl_conversions/pcl_conversions.h:70,
                 from /home/shiyunyan/catkin_ws/src/Teach-Repeat-Replan/local_replanner/local_perception/include/local_perception/sdf_map.h:15,
                 from /home/shiyunyan/catkin_ws/src/Teach-Repeat-Replan/local_replanner/local_perception/src/sdf_map.cpp:1:
/usr/include/pcl-1.10/pcl/common/io.h:65:39: note: candidate: pcl::getFieldIndex(const pcl::PCLPointCloud2&, const string&)::<lambda(int)>
         [&field_name](const auto field) { return field.name == field_name; });
                                       ^
/usr/include/pcl-1.10/pcl/common/io.h:65:39: note:   no known conversion for argument 1 from ‘const pcl::PCLPointField’ to ‘int’
make[2]: *** [Teach-Repeat-Replan/local_replanner/local_perception/CMakeFiles/local_map.dir/build.make:63：Teach-Repeat-Replan/local_replanner/local_perception/CMakeFiles/local_map.dir/src/sdf_map.cpp.o] 错误 1
make[1]: *** [CMakeFiles/Makefile2:2217：Teach-Repeat-Replan/local_replanner/local_perception/CMakeFiles/local_map.dir/all] 错误 2
make: *** [Makefile:141：all] 错误 2
Invoking "make -j1" failed
请问我该如何解决呢？谢谢您～

bigsuperZZZX, 2021-12-28:
目测pcl版本问题，建议搞个虚拟机纯净系统测试编译，控制变量找问题

shiyunyan7, 2021-12-28:
谢谢您的回复！

bigsuperZZZX, 2022-01-14:
没啥更多问题的话我就关了这个issue了

closed 2022-01-14 by bigsuperZZZX