HKUST-Aerial-Robotics/Teach-Repeat-Replan #56: How to tune controller parameters?

dssdyx, 2022-03-23:
I modified n3ctrl to adjust px4 FCU. It works well in gazebo, but it cannot work as well as SITL in real experiment. I think it's the parameter problem. Could you tell me how to tune this controller parameters,which parameters are important and which tool you use for tuning? Hope for your advice!

bigsuperZZZX, 2022-03-23:
You can adjust the gain parameters in the [configure file](https://github.com/HKUST-Aerial-Robotics/Teach-Repeat-Replan/blob/experiment/onboard_computer/controller/n3ctrl/config/ctrl_param_fpv.yaml).
n3ctrl should work well with px4 FCU.

dssdyx, 2022-03-23:
> You can adjust the gain parameters in the [configure file](https://github.com/HKUST-Aerial-Robotics/Teach-Repeat-Replan/blob/experiment/onboard_computer/controller/n3ctrl/config/ctrl_param_fpv.yaml). n3ctrl should work well with px4 FCU.

Thank you for your answer. I know these gain parameters, and could you tell me the steps and some experiments about tuning them? I have destroyed my vehicle too many times. :(

bigsuperZZZX, 2022-03-25:
Could you describe the drone reactions to your parameters? I have no idea if you don't tell me anything about your drone.

dssdyx, 2022-03-26:
> Could you describe the drone reactions to your parameters? I have no idea if you don't tell me anything about your drone.

OK. I use a 360mm UAV to test. I first use standard trajectory like circle and eight to try fitness. It's ok. But,when I use some planner algorithm, it's trajectories are always overshoot or slow respond in position. I try to adjust track parameters like kp,kv, and observe the roll and pitch respond in PX4 QGC. It seems that however I adjust these parameters, the curves of setpoint and current values in roll and pitch have no change. Should I observe pos and vel curves to compare? If it is yes, which one should first adjust?

shubham-shahh, 2022-04-29:
Hi, @dssdyx how did you convert the control commands from n3ctrl to px4 compatible messages?
thanks

shubham-shahh, 2022-04-29:
> OK. I use a 360mm UAV to test. I first use standard trajectory like circle and eight to try fitness. It's ok. But,when I use some planner algorithm, it's trajectories are always overshoot or slow [...]

Hi, This could also be because your PX4 controller gains are not tuned for aggressive flights, you can share logs for further analysis

thanks

dssdyx, 2023-09-26:
> Hi, This could also be because your PX4 controller gains are not tuned for aggressive flights, you can share logs for further analysis
>
> thanks

PX4 supports SET ATTITUDE TARGET mavlink to control fcu. I use mavros and publish /mavros/setpoint_raw/attitude topic to set desired q and thrust. You can search Fast Drone 250 project to learn how to convert.

shubham-shahh, 2023-09-26:
> > > > Could you describe the drone reactions to your parameters? I have no idea if you don't tell me anything about your drone.
> > > 
> > > 
> > > OK. I use a 360mm UAV to test. I first use standard trajectory like circle and eight to try fitness. It's ok. But,when I use some planner algorithm, it's trajectories are always overshoot or slow respond in position. I try to adjust track parameters like kp,kv, and observe the roll and pitch respond in PX4 QGC. It seems that however I adjust these parameters, the curves of setpoint and current values in roll and pitch have no change. Should I observe pos and vel curves to compare? If it is yes, which one should first adjust?
> > 
> > Hi, This could also be because your PX4 controller gains are not tuned for aggressive flights, you can share logs for further analysis
> > 
> > thanks
> 
> Sorry forgetting focusing on this question.
> PX4 supports SET ATTITUDE TARGET mavlink to control fcu. I use mavros and publish /mavros/setpoint_raw/attitude topic to set desired q and thrust. You can search Fast Drone 250 project to learn how to convert. 
> If hoping for a good performance on PX4 fcu, a stable and sensitive attitude controller and rate controller pid param is necessary. 
> Then, onboard code including localization and tracking controller, must be stable and not result in divergency. You should confirm your onboard pc can hold these threads. Tracking controller, focusing on position PI and velecity P. Actually, if tracking performance is not good, I think just increase pos P and vel P, no let drone vibrate.

Hi, thanks for a detailed response, do you have any resources to understand how can we control "thrust" with SET_ATTITUDE_TARGET message? Also have you tried using mavros_controllers/geometric controller?

Thanks

bigsuperZZZX, 2023-10-08:
[This file](https://github.com/ZJU-FAST-Lab/Fast-Drone-250/blob/8ff64274b161bb829653faf5946656658e7e2c9f/src/realflight_modules/px4ctrl/src/controller.cpp#L36) contains how we control thrust.