HKUST-Aerial-Robotics/Teach-Repeat-Replan #57: undefined reference to `mocka::OperationalLogic::RegisterSDK::id'

RotKai, 2022-03-30:
When I do catkin_make for the file, this error comes up. 
'''
/usr/bin/ld: CMakeFiles/state_machine_node.dir/node/statemachine_node.cpp.o: in function `void mocka::OperationalLogic::addEvent<mocka::RC::EVENT>(mocka::RC::EVENT const&)':
statemachine_node.cpp:(.text+0x181): undefined reference to `mocka::OperationalLogic::RegisterSDK::id'
'''
I am working on ubuntu 20.04. I was trying to use different versions of gcc, g++, and gfortran but all didn't work. Is there a solution for this?

bigsuperZZZX, 2022-07-01:
Sorry, I have no idea where this error comes from.

closed 2022-07-01 by bigsuperZZZX

Biude, 2023-02-06:
> When I do catkin_make for the file, this error comes up. ''' /usr/bin/ld: CMakeFiles/state_machine_node.dir/node/statemachine_node.cpp.o: in function `void [...]

Biude, 2023-02-06:
我也遇到了同样的问题。
系统版本是ubuntu20.04   ros noetic

bigsuperZZZX, 2023-02-08:
提示未定义的“id”其实定义在[这里](https://github.com/HKUST-Aerial-Robotics/Teach-Repeat-Replan/blob/1e7ab15e7d37314f67f9eca79ef6005fa9fd1c87/simulation/state_machine/include/statemachine.hpp#L72)，至于为啥找不到，我也想不明白为什么。
毕竟这个软件是基于16.04写的，可能20.04确实是太新了

Biude, 2023-02-08:
好的，谢谢。那我试试其他系统版本