HKUST-Aerial-Robotics/Teach-Repeat-Replan #59: Ubuntu 18.04 OOQP make error

YouShaoze, 2022-06-29:
g++ -O  -Isrc/LinearSolvers/  -Isrc/SparseLinearAlgebra/  -Isrc/Ampl/  -Isrc/Vector/  -Isrc/PetscLinearAlgebra/  -Isrc/QpBound/  -Isrc/Readers/  -Isrc/Svm/  -Isrc/QpSolvers/  -Isrc/Abstract/  -Isrc/QpGen/  -Isrc/Utilities/  -Isrc/Huber/  -Isrc/Mex/  -Isrc/QpExample/  -Isrc/CInterface/  -Isrc/DenseLinearAlgebra/  -Isrc/LinearSolvers/Ma57Solver/  -Isrc/LinearSolvers/Ma27Solver/ -DPACKAGE_NAME=\"\" -DPACKAGE_TARNAME=\"\" -DPACKAGE_VERSION=\"\" -DPACKAGE_STRING=\"\" -DPACKAGE_BUGREPORT=\"\" -DPACKAGE_URL=\"\" -DSTDC_HEADERS=1 -DHAVE_GETRUSAGE=1   -o qpbound-dense-gondzio.exe \
	src/QpBound/QpBoundDenseGondzioDriver.o src/QpBound/QpBoundDenseLinsys.o src/QpBound/QpBoundDense.o ./lib/libooqpgondzio.a ./lib/libooqpbound.a ./lib/libooqpdense.a ./lib/libooqpbase.a \
	/home/ww/Depend/blas/BLAS-3.10.0/libfblas.a -L/usr/lib/gcc/x86_64-linux-gnu/7 -L/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu -L/usr/lib/gcc/x86_64-linux-gnu/7/../../../../lib -L/lib/x86_64-linux-gnu -L/lib/../lib -L/usr/lib/x86_64-linux-gnu -L/usr/lib/../lib -L/usr/lib/gcc/x86_64-linux-gnu/7/../../.. -lgfortran -lm -lquadmath 
./lib/libooqpdense.a(DeSymPSDSolver.o): In function `DeSymPSDSolver::matrixChanged()':
DeSymPSDSolver.C:(.text+0x4e): undefined reference to `dpotrf_'
./lib/libooqpdense.a(DeSymPSDSolver.o): In function `DeSymPSDSolver::solve(OoqpVector&)':
DeSymPSDSolver.C:(.text+0xe2): undefined reference to `dpotrs_'
collect2: error: ld returned 1 exit status
src/QpBound/MakefileTargets.inc:14: recipe for target 'qpbound-dense-gondzio.exe' failed
make: *** [qpbound-dense-gondzio.exe] Error 1
从git上拉取的源码编译或者是从此包中解压的编译都出现了这个情况.....一直没有找到解决方法

bigsuperZZZX, 2022-07-01:
OOQP安装起来挺麻烦的，需要想办法单独安装

YouShaoze, 2022-07-05:
具体前置的配置环境还需要什么呢? 我blas是3.10,lapack也是3.10,MA27也安装了,gcc我5和7都试过了...也是会出现这个问题...全网也找不到跟我同样问题的 ToT

closed 2022-08-04 by bigsuperZZZX