HKUST-Aerial-Robotics/Teach-Repeat-Replan #62: catkin_make failed

YouShaoze, 2022-12-01:
[ 90%] Building CXX object Teach-Repeat-Replan/simulation/mission_template/CMakeFiles/mission_template.dir/src/mission_template.cpp.o
g++-7: error: unrecognized command line option ‘-std=c++11 ’; did you mean ‘-std=c++11’?
Teach-Repeat-Replan/simulation/mission_template/CMakeFiles/mission_template.dir/build.make:62: recipe for target 'Teach-Repeat-Replan/simulation/mission_template/CMakeFiles/mission_template.dir/src/mission_template.cpp.o' failed
make[2]: *** [Teach-Repeat-Replan/simulation/mission_template/CMakeFiles/mission_template.dir/src/mission_template.cpp.o] Error 1
CMakeFiles/Makefile2:6405: recipe for target 'Teach-Repeat-Replan/simulation/mission_template/CMakeFiles/mission_template.dir/all' failed
make[1]: *** [Teach-Repeat-Replan/simulation/mission_template/CMakeFiles/mission_template.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 94%] Built target decomp_rviz_plugins
[ 97%] Built target tr_node
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j12 -l12" failed

YouShaoze, 2022-12-01:
ww@ww:~/TRR$ /usr/bin/gcc --version
gcc (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

ww@ww:~/TRR$ sudo update-alternatives --config gcc
There are 3 choices for the alternative gcc (providing /usr/bin/gcc).

  Selection    Path            Priority   Status
------------------------------------------------------------
  0            /usr/bin/gcc-5   60        auto mode
  1            /usr/bin/gcc-5   60        manual mode
* 2            /usr/bin/gcc-7   50        manual mode
  3            /usr/bin/gcc-8   10        manual mode

ww@ww:~/TRR$ gfortran --version
GNU Fortran (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

ww@ww:~/TRR$ g++ --version
g++ (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

bigsuperZZZX, 2023-01-29:
Something goes wrong with the g++ compiler. 
You could delete the white space after the ‘-std=c++11’ flag or just delete the whole ‘-std=c++11’ flag.