HKUST-Aerial-Robotics/Teach-Repeat-Replan #67: Pointcloud orientation problem

enordco, 2023-11-04:
Hi,
Im using realsense d435i and facing an issue of pointcloud in z axis. I am using depth topic of realsense.
I am giving my odometry topic whose forward is z, and also giving my odometry pose topic.
[image]
Please help

bigsuperZZZX, 2023-11-11:
1. Check [pose_type_ ](https://github.com/HKUST-Aerial-Robotics/Teach-Repeat-Replan/blob/98505a7f74b13c8b501176ff838a38423dbef536/local_replanner/local_perception/src/sdf_map.cpp#L670) to see which  callback function will be used. _SDFMap::depthPoseCallback()_ is recommended.
2. If you use _SDFMap::depthOdomCallback()_ instead, please add a proper rotation when [assigning the value from odom to camera_q_](https://github.com/HKUST-Aerial-Robotics/Teach-Repeat-Replan/blob/98505a7f74b13c8b501176ff838a38423dbef536/local_replanner/local_perception/src/sdf_map.cpp#L1246C3-L1246C13).

enordco, 2023-11-11:
> 1. Check [pose_type_ ](https://github.com/HKUST-Aerial-Robotics/Teach-Repeat-Replan/blob/98505a7f74b13c8b501176ff838a38423dbef536/local_replanner/local_perception/src/sdf_map.cpp#L670) to see which [...]

Thank you for the reply!!
As I already mentioned that I am using odometry(open_vins), and all the message type are corrected i have checked it again and Forward of odometry is x but pointcloud is facing z axis. 
@bigsuperZZZX

bigsuperZZZX, 2024-04-22:
You can add a rotation matrix like this https://github.com/ZJU-FAST-Lab/ego-planner/blob/0835f963dcfadce156dbec8ab66b93930955a87c/src/planner/plan_env/src/grid_map.cpp#L91