HKUST-Aerial-Robotics/Teach-Repeat-Replan #68: 代码能否适配ubuntu20.04？

INDEX108, 2024-11-12:
我在ubuntu20.04下编译，主要的PCL版本问题解决后 仍然报错
[ 97%] Linking CXX executable /home/index108/Teach-Repeat-Replan/devel/lib/trr_global_planner/click_obstacle_generation
[ 98%] Linking CXX executable /home/index108/Teach-Repeat-Replan/devel/lib/trr_global_planner/random_complex
[ 98%] Built target click_obstacle_generation
[ 98%] Built target random_complex
[100%] Linking CXX executable /home/index108/Teach-Repeat-Replan/devel/lib/trr_global_planner/tr_node
/usr/bin/ld: 找不到 -lgfortran
/usr/bin/ld: 找不到 -lgfortran
collect2: error: ld returned 1 exit status
make[2]: *** [Teach-Repeat-Replan/global_planner/CMakeFiles/tr_node.dir/build.make:483：/home/index108/wyx/Teach-Repeat-Replan/devel/lib/trr_global_planner/tr_node] 错误 1
make[1]: *** [CMakeFiles/Makefile2:4275：Teach-Repeat-Replan/global_planner/CMakeFiles/tr_node.dir/all] 错误 2
make: *** [Makefile:141：all] 错误 2
Invoking "make -j16 -l16" failed

closed 2024-11-13 by INDEX108

yangjiawen2001, 2024-11-16:
兄弟你问题是怎么解决的

yangjiawen2001, 2024-11-16:
您好，我最近也碰到了这个问题，想请教下您是怎么解决的

INDEX108, 2024-11-16:
建立软链接，将现有的***gfortan.so.*链接到***gfortan.so，就能找到了具体名字和位置用locate找一下

> 您好，我最近也碰到了这个问题，想请教下您是怎么解决的

INDEX108, 2024-11-16:
建立软链接，将现有的***gfortan.so.*链接到*gfortan.so，就能找到了具体名字和位置用locate找一下

> 您好，我最近也碰到了这个问题，想请教下您是怎么解决的

INDEX108, 2024-11-16:
> 您好，我最近也碰到了这个问题，想请教下您是怎么解决的

建立软链接，将现有的xxxgfortan.so.x链接到xxxgfortan.so，就能找到了具体名字和位置用locate找一下

yangjiawen2001, 2024-11-16:
可以说的更详细些吗，我用 locate 找到了四个文件，不知道下一步应该怎么做
```
locate -r libgfortran.so.*
/usr/lib/gcc/x86_64-linux-gnu/8/libgfortran.so
/usr/lib/gcc/x86_64-linux-gnu/9/libgfortran.so
/usr/lib/x86_64-linux-gnu/libgfortran.so.5
/usr/lib/x86_64-linux-gnu/libgfortran.so.5.0.0
```

yangjiawen2001, 2024-11-16:
> > 您好，我最近也碰到了这个问题，想请教下您是怎么解决的
> 
> 建立软链接，将现有的xxxgfortan.so.x链接到xxxgfortan.so，就能找到了具体名字和位置用locate找一下

可以说的更详细些吗，我用 locate 找到了 4 个文件，不知道下一步应该怎么做
```
$ locate -r libgfortran.so.*
/usr/lib/gcc/x86_64-linux-gnu/8/libgfortran.so
/usr/lib/gcc/x86_64-linux-gnu/9/libgfortran.so
/usr/lib/x86_64-linux-gnu/libgfortran.so.5
/usr/lib/x86_64-linux-gnu/libgfortran.so.5.0.0
```

INDEX108, 2024-11-16:
.9/.8那个就可以

yangjiawen2001, 2024-11-16:
软链接是要这样建立吗
`sudo ln -s /usr/lib/x86_64-linux-gnu/libgfortran.so.5 /usr/lib/gcc/x86_64-linux-gnu/9/libgfortran.so`

INDEX108, 2024-11-16:
是的 我引得.9

yangjiawen2001, 2024-11-16:
感谢您的回复，我使用了另一种方法，在 src/Teach-Repeat-Replan/global_planner/CMakeLists.txt 里使用 `link_directories()` 设置链接库的路径：
`link_directories(/usr/lib/gcc/x86_64-linux-gnu/9)`